balena-io/etcher #3675: Only seeing white screen on NixOS

ertw, 2022-01-08:
- **Etcher version:** 1.6.0
- **Operating system and architecture:** NixOS 21.11 Linux 5.15.11 x86_64
- **Image flashed:** NA
- **Do you see any meaningful error information in the DevTools?**

in DevTools:

```
index.css:1 Failed to load resource: net::ERR_FILE_NOT_FOUND
gui.js:7117 Uncaught TypeError: Cannot read properties of undefined (reading 'app')
    at appPath (gui.js:7117)
    at Object.exports.CODE_TO_ERRNO.E2BIG (gui.js:7119)
    at __webpack_require__ (gui.js:72085)
    at Object.__webpack_modules__.constructor (gui.js:5691)
    at __webpack_require__ (gui.js:72085)
    at Object.__webpack_modules__.DiskWrapper.DiskWrapper (gui.js:5686)
    at __webpack_require__ (gui.js:72085)
    at Object.<anonymous> (gui.js:5587)
    at __webpack_require__ (gui.js:72085)
    at Object.defineProperty.value (gui.js:5401)
```

and in the terminal:

```
[Axios v1.6.0] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.6.0] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.6.0] Transitional option 'silentJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[50630:0108/074655.482195:ERROR:sandbox_linux.cc(374)] InitializeSandbox() called with multiple threads in process gpu-process.
(node:50593) electron: The default of nativeWindowOpen is deprecated and will be changing from false to true in Electron 15.  See https://github.com/electron/electron/issues/28511 for more information.
(Use `electron --trace-warnings ...` to show where the warning was created)
ready-to-show: 245.009ms
(node:50593) Warning: No such label 'ready-to-show' for console.timeEnd()

(electron:50593): Gtk-WARNING **: 07:47:10.865: Could not find the icon 'user-home-symbolic-ltr'. The 'hicolor' theme
was not found either, perhaps you need to install it.
You can get a copy from:
	http://icon-theme.freedesktop.org/releases
```

I've been happily using Etcher on NixOS for years, but in some OS update in the last few months, this broke. Etcher now presents only as a white screen when launching. I'd appreciate any help in troubleshooting so I can get back to flashing images from my main OS, although I have a workaround of just booting into a different OS where Etcher works flawlessly.

Example of what I see when running Etcher:
[image]

thundron, 2022-01-08:
@ertw could you please try with the latest etcher v1.7.3?

ertw, 2022-01-09:
> @ertw could you please try with the latest etcher v1.7.3?

Thanks, I will close this issue for now, and reopen if still present in the latest version.

closed 2022-01-09 by ertw