balena-io/etcher #3677: Cloudsmith DEB Repository not working

token47, 2022-01-10:
I just tried to update packages and got:

$ sudo apt update
[...]
Err:14 https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu focal InRelease
  402  Payment Required [IP: 52.85.208.44 443]
[...]
E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu/dists/focal/InRelease  402  Payment Required [IP: 52.85.208.44 443]
E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu focal InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

Also, the shell script that is used to install the repo is also not available:

$ wget https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh
--2022-01-10 09:45:45--  https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh
Resolving dl.cloudsmith.io (dl.cloudsmith.io)... 52.85.208.53, 52.85.208.23, 52.85.208.44, ...
Connecting to dl.cloudsmith.io (dl.cloudsmith.io)|52.85.208.53|:443... connected.
HTTP request sent, awaiting response... 402 Payment Required
2022-01-10 09:45:46 ERROR 402: Payment Required.

RHITNL, 2022-01-10:
Hi,

just noticed the same for my Debian 11 setup.
I used the latest deb on https://github.com/balena-io/etcher/releases and installed it with `sudo apt install ./balena-etcher-electron_1.7.3_amd64.deb`

Regards,

Ralph

mcraa, 2022-01-10:
duplicate of #3672

closed 2022-01-10 by mcraa