balena-io/etcher #3678: it dosent work in zorin

vatsalparmar2009, 2022-01-13:
i want to install balena etcher in zorin os 16 by deb install so i coped command of curl and bash after it run it tell me this error
Executing the  setup script for the 'balena/etcher' repository ...

   OK: Checking for required executable 'curl' ...
   OK: Checking for required executable 'apt-get' ...
   OK: Detecting your OS distribution and release using system methods ...
 ^^^^: ... Detected/provided for your OS/distribution, version and architecture:
 >>>>:
 >>>>: ... distro=zorin  version=16  codename=focal  arch=x86_64  
 >>>>:
   OK: Checking for apt dependency 'apt-transport-https' ...
   OK: Checking for apt dependency 'ca-certificates' ...
   OK: Checking for apt dependency 'gnupg' ...
   OK: Importing 'balena/etcher' repository GPG key into apt ...
 NOPE: Checking if upstream install config is OK ...
 ^^^^: ... Failed to fetch configuration for your OS distribution release/version.
 >>>>:
 >>>>: It looks like we don't currently support your distribution release and
 >>>>: version. This is something that we can fix by adding it to our list of
 >>>>: supported versions (see contact us below), or you can manually override
 >>>>: the values below to an equivalent distribution that we do support:
 >>>>:
 >>>>: ... distro=zorin  version=16  codename=focal  arch=x86_64  
 >>>>:
 >>>>: You can force this script to use a particular value by specifying distro,
 >>>>: version, or codename via environment variable. E.g., to specify a distro
 >>>>: such as Ubuntu/Xenial (16.04), use the following:
 >>>>:
 >>>>:   <curl command> | distro=ubuntu version=16.04 codename=xenial sudo -E bash
 >>>>:
 !!!!: Oh no, your setup failed! :-( ... But we might be able to help. :-)
 !!!!: 
 !!!!: You can contact balena for further assistance.
 !!!!: 
 !!!!: URL: https://github.com/balena-io/etcher
 !!!!: 
 !!!!: Last 20 log lines from .csm_setup_cTo7bMx2N (might not be errors, nor even relevant):
 !!!!:
 !!!!: > Warning: apt-key output should not be parsed (stdout is not a terminal)
 !!!!: > OK

and then stop the script

lurch, 2022-01-13:
Duplicate of #3600 ?

vatsalparmar2009, 2022-01-13:
sir it is installed

vatsalparmar2009, 2022-01-13:
sir it installed 
root@vatsal:/home/vatsal# apt-get install -y apt-transport-https
Reading package lists... Done
Building dependency tree       
Reading state information... Done
apt-transport-https is already the newest version (2.0.6).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@vatsal:/home/vatsal# apt-get install -y debian-archive-keyring  
Reading package lists... Done
Building dependency tree       
Reading state information... Done
debian-archive-keyring is already the newest version (2019.1ubuntu2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@vatsal:/home/vatsal# apt-get install -y debian-keyring
Reading package lists... Done
Building dependency tree       
Reading state information... Done
debian-keyring is already the newest version (2020.03.24).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@vatsal:/home/vatsal#