balena-io/etcher #3680: Flash ubuntu-20.04.3-desktop-amd64.iso failed for "There are some data after the end of the payload data". It is a warning in 7-zip Test archive.

winteryu, 2022-01-15:
- **Etcher version:** 1.7.3
- **Operating system and architecture:** Windows10, x64
- **Image flashed:** ubuntu-20.04.3-desktop-amd64.iso
- **Do you see any meaningful error information in the DevTools?** It has a warning in Test archive: There are some data after the end of the payload data

winteryu, 2022-01-15:
It has a warning in Test archive in 7-zip: There are some data after the end of the payload data.

Well2333, 2022-01-16:
same. Beside, It's seems destroyed both my sd card and usb disk while I try to flashing. Both of them cant be deteced in windows and DiskGenius.

Arrendy, 2022-01-20:
Same version of Etcher/Windows/Ubuntu fails for me. Formatted usb and no change. Never had issue with etcher before

CrossHusky, 2022-02-16:
Could this be related to the iso itself?

reesezxf, 2022-06-16:
I have the same question, windows 10, balenaEtcher-Portable-1.5.115.exe, ubuntu-20.04.4-desktop-amd64.iso. I trid twice,all failed.

aashu, 2023-03-05:
Did you guys found any solutions yet? I am facing same issue.