balena-io/etcher #3687: The capacity of Micro SD card is displayed incorrectly.

BahuangShanren, 2022-01-25:
- **Etcher version:** v1.7.3
- **Operating system and architecture:** Windows11,amd64
- **Image flashed:** [FydeOS_for_PC_v13.1-stable.img.xz](https://fydeos.com/download/pc/intel-hd) 
- **Do you see any meaningful error information in the DevTools?** None

This is just a Micro SD card with 32GB capacity (actually only 29.1GB). 

[image]

But etcher indicates that it has a capacity of 2.2TB, which is puzzling.

[image]

I still perform flash operations to see what happens.As a result, etcher prompts that the flash failed, and then the capacity of Micro SD becomes 2045.12GB.

[image]

lurch, 2022-01-25:
It might be worth trying a different USB card-reader, in case there's some kind of incompatibility with your new SD-card and your old card-reader? :shrug:

BahuangShanren, 2022-01-25:
> It might be worth trying a different USB card-reader, in case there's some kind of incompatibility with your new SD-card and your old card-reader? 🤷

Maybe you're right. I have tried to use the card reader to carry other SD cards, and their capacity is displayed correctly.
But the capacity of this Kingston 32GB micro SD card has been changed, and I can't restore it. So I can't verify whether the cause of the problem is what you said.

lurch, 2022-01-25:
I was assuming that perhaps a different card-reader would be able to read the "correct" size? :shrug:

BahuangShanren, 2022-01-25:
> I was assuming that perhaps a different card-reader would be able to read the "correct" size? 🤷

closed 2022-01-25 by BahuangShanren