balena-io/etcher #3689: usb pen damaged

frankbald, 2022-01-27:
- **1.73** 
- **windows 10** 
- **linux-lite-5.4-64bit.iso** 

I just tried to flash this iso on my 16 gb usb crucial pen, now I can't see it anymore from windows explorer. Tried to rebuild the partition with disk management, nothing happened, then AOMEI partition assistant, the same, what can I do?

lurch, 2022-01-27:
Does https://www.balena.io/blog/did-etcher-break-my-usb-sd-card/ help?

frankbald, 2022-01-27:
> Does https://www.balena.io/blog/did-etcher-break-my-usb-sd-card/ help?

thanks, long discussion but at the end I found the solution. Clearly this software has a huge bugs seeing all the comments

lurch, 2022-01-27:
I think the "bugs" are with the way that Windows copes with non-standard partitioning schemes, rather than with Etcher? :man_shrugging:

frankbald, 2022-01-27:
> I think the "bugs" are with the way that Windows copes with non-standard partitioning schemes, rather than with Etcher? 🤷‍♂️

maybe, but first of all windows takes the majority of the os market share, so if you develop something you have to take into account what's best for who is using Etcher on Windows. Second and most important I've used disk management and aomei partition manager, and it's really weird that both of them were unable to reverse the stuck state of the pen

mcraa, 2022-01-27:
Can you share what was the solution, maybe from the solution backtrack the details of the problem, and share those too?
So others can use your comments if they have the same issue.

Would you like to open issues or support tickets at Microsoft and aomei that their software was not helpful with a pen drive that you were able to fix in a different way? Maybe share the replies from there too so others can use your comments if they have the same issue.

Otherwise, we can close this issue,
Thanks

frankbald, 2022-01-27:
i've used the following commands on powershell opened in administrator mode:

DISKPART> clean

DiskPart has encountered an error: Access is denied.
See the System Event Log for more information.

DISKPART> convert gpt

DiskPart successfully converted the selected disk to GPT format.

DISKPART> create partition primary

DiskPart succeeded in creating the specified partition.

DISKPART> format quick fs=ntfs

  100 percent completed

DiskPart successfully formatted the volume.

frankbald, 2022-01-28:
> I'm not directly involved with the Etcher project, but just to clarify...
> 
> > you have to take into account what's best for who is using Etcher on Windows.
> 
> The "issue" isn't caused by Etcher itself, but with the contents of your `linux-lite-5.4-64bit.iso` file. (i.e. I suspect using Win32DiskImager or RPi Imager would cause exactly the same problems)

nope, it's not caused by the iso, and then if you go to read comments on this page https://www.balena.io/blog/did-etcher-break-my-usb-sd-card/ you can see that there are a lot of people using Etcher with all kinds of iso having this problem, so yeah, it's a Etcher bug, I don't understand why some people are taking this as a personal offense, user feedback should be useful to improve software

lurch, 2022-01-28:
I wasn't taking personal offence, I don't care if you use Etcher or not, I was just trying to clear up confusion.

There's more info about the confusion "hybrid Linux ISOs" can cause [here](https://github.com/balena-io/etcher/blob/master/FAQ.md#how-do-i-flash-ubuntu-isos).
And see also [this discussion](https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive) (and the [massive thread](https://github.com/pbatard/rufus/issues/313#issuecomment-322045700) that it links to) for an example of where people prematurely blame the disk-flashing software for being faulty, when it's not actually the software's fault.

frankbald, 2022-01-28:
and even better, why the developers don't put inside the tool a formatter option to restore the memory device to a default fat32 partition?

lurch, 2022-01-28:
> and even better, why the developers don't put inside the tool a formatter option to restore the memory device to a default fat32 partition?

I've no idea why Etcher hasn't yet added an option to do that (see #273 ), but https://github.com/raspberrypi/rpi-imager/ **does** have the option to do that.

hanoied, 2026-02-13:
It's a real shame I didn't know or even think to research this potential problem before unintentionally turning my 128GB USB drive into a 5MB piece of junk while trying to flash Linux Mint ISO onto it as a bootable drive.... Largely my own fault for attempting to follow step-by-step instructions without having any real idea what I was doing, but really disappointing too to see that over 4 years on and the OP's sensible suggestion that a helpful disclaimer be included highlighting what seems to be a common occurrence doesn't seem to have been taken on board or implemented. 

Could somebody kindly point me in the direction of a detailed fix so I might be able to restore my USB drive to its original capacity?

Thanks! :o)