balena-io/etcher #3692: Etcher - can't change the language - HOW can i change the interface language ?

heidiwenger, 2022-01-30:
From english to something else.  I'm using the Linux AppImage.  

On the Settings menu ("gear icon") there are only two options:

"Anonymously report errors and usage statistics to balena.io"
and 

"Auto-updates enabled". 

Why is there nothing else? How or where can i change the language of the interface?

lurch, 2022-01-30:
You can't yet. See #308

heidiwenger, 2022-01-30:
> You can't yet. See #308

Oh, ok. But this has been on at least since 2016?? Is there some obstacle preventing the language change or what is it? Why is it taking so long? 

Thanks.

thundron, 2022-02-01:
@heidiwenger virtually, nothing! any community member can just pick it up and implement it with a bit of communication with balena to check the direction to go with. it's the beauty of OSS ❤️

btwise, 2022-02-12:
> From english to something else. I'm using the Linux AppImage.
>
> On the Settings menu ("gear icon") there are only two options:
>
> "Anonymously report errors and usage statistics to balena.io" and [...]

Modify the source code and change it to your own language, which I have done!

RoderickQiu, 2022-07-16:
> 

If I'm fortunate enough to have enough time in recent time, I think I will be interested to contribute an i18n part for it.