balena-io/etcher #3701: Failed to flash image built by balena-rockpi

sprhawk, 2022-02-14:
- **Etcher version:**  1.7.3
- **Operating system and architecture:**  Debian Buster
- **Image flashed:**  built by [`balena-rockpi`](https://github.com/balena-os/balena-rockpi)
- **Do you see any meaningful error information in the DevTools?**  

```
gui.js:46 {"stack":"Error: The elevated process died unexpectedly\n    at createError (file:///tmp/.mount_balenaDFEDEY/resources/app/generated/gui.js:46:119883)\n    at Object.createUserError (file:///tmp/.mount_balenaDFEDEY/resources/app/generated/gui.js:46:120096)\n    at file:///tmp/.mount_balenaDFEDEY/resources/app/generated/gui.js:351:4996\n    at async withTmpFile (file:///tmp/.mount_balenaDFEDEY/resources/app/generated/gui.js:1:344288)\n    at async elevateCommand (file:///tmp/.mount_balenaDFEDEY/resources/app/generated/gui.js:351:4227)\n    at async Server.<anonymous> (file:///tmp/.mount_balenaDFEDEY/resources/app/generated/gui.js:351:8331)","message":"The elevated process died unexpectedly","description":"The process error code was 126","report":false,"image":"balena-image-flasher-rockpi-4b-rk3399-2022-02-14.img"}
```

[image]

I can use this version of ether to flash official downloaded images.

mcraa, 2022-02-14:
Hi @sprhawk 

> built by `balena-rockpi`

you mean you build the image yourself from the repo?
I yes can you share the details (flags, env, etc)

From the rockpi image downloaded from balena dashboard I can't reproduce the issue.
In your devtools logs `error code 126` most probably ENOKEY

sprhawk, 2022-02-15:
> you mean you build the image yourself from the repo? I yes can you share the details (flags, env, etc)

Yes, it is. I just use the default local.conf generated by `barys`

> From the rockpi image downloaded from balena dashboard I can't reproduce the issue. In your devtools logs `error code 126` most probably ENOKEY

If it is, what `KEY` do I need ?

sprhawk, 2022-02-16:
> Sorry, @sprhawk, my mistake I was wondering too how could etcher cause ENOKEY, please disregard.
> 
> ...
> [126](https://www.gnu.org/software/bash/manual/html_node/Exit-Status.html) should mean "simply" permission error. Did you get the sudo prompt prior to flashing? Can you check the permissions on your build output img?
Yes, after I enter my sudo password, it report failure. All the process is same as using the images downloaded from balena.io
the file permission is `-rw-r--r--`