balena-io/etcher #3706: Etcher won't open with "Out of memory" error

Occhioverde, 2022-02-20:
- **Etcher version:** 1.7.3
- **Operating system and architecture:** Manjaro Linux 21.2.3 with Kernel 5.16.7
- **Image flashed:** N/A
- **Do you see any meaningful error information in the DevTools?** N/A

As per the title, today I tried to open Etcher to write an image on my SD card, but the program crashed after a couple of minutes of waiting without seeing anything appearing on-screen.

Below, you can find the full output printed by Etcher when started from the command line:
```
❯ etcher
interface 'wl_output' has no event 4
tcmalloc: large alloc 1073741824 bytes == 0x90a408b6000 @  0x564d9eece596 0x7f8b1de67047
[1787:0220/214402.101096:FATAL:memory_linux.cc(37)] Out of memory.
fish: Job 1, 'etcher' terminated by signal SIGTRAP (Trace or breakpoint trap)
```

Occhioverde, 2022-03-05:
The problem seems to be present only on the package provided by Manjaro's community repo since I managed to solve it by installing the etcher-bin AUR version.
I'll close this issue as the error wasn't caused directly by Etcher.

closed 2022-03-05 by Occhioverde

BrunoBernardino, 2022-11-04:
If someone else ends up here like I did (I was on Ubuntu 22.10), I solved it by running the command (or AppImage, I had the same problem with it) with `sudo` (the error still showed in the terminal, but the app opened and it all worked).