balena-io/etcher #3707: MacOS download button does not work

justinlawrence, 2022-02-21:
See [www.](https://www.balena.io/etcher/). If you click on "Download for MacOS", it shows a 404 page. Thanks.

mcraa, 2022-02-21:
@justinlawrence thanks, we know, the CI for the last version uploaded only the windows files as mentioned [here](https://github.com/balena-io/etcher/pull/3697#issuecomment-1046796235) probably due to network errors.
We are working on it.

justinlawrence, 2022-02-21:
> We are working on it.

@mcraa brilliant, thanks for the speedy response!

mcraa, 2022-02-21:
until the site is fixed, the latest version is available directly by github releases
https://github.com/balena-io/etcher/releases/tag/v1.7.6

mcraa, 2022-02-22:
with version 1.7.7 the site is updated correctly

closed 2022-02-22 by mcraa

justinlawrence, 2022-02-22:
Thanks guys!