balena-io/etcher #3714: Community builds

mcraa, 2022-02-28:
The official build of balenaEtcher has Linux, Windows, and Mac versions, but may not cover all the possible combinations of operating systems and underlying architectures.
Embracing the power of the Opensource community, you can create a build if you feel one is missing, and share it here.

- MacOS - Aarch64 {{gh-handle}} {{link coming soon}} 
- etc.

Simply send us a message or comment with the link and description of your hosted build output, and we will add it to the list.

> Note:  These versions are not built, maintained, or supervised by balena, so proceed with caution. 

> Tip: An easy way of hosting a build is to create a GitHub repo, then create a Release in the repo, and upload the files there. You may put the description of your build in the README of the repo.

thedocbwarren, 2022-03-19:
Updated build for Apple Silicon v1.7.8

https://github.com/Augmentedjs/balena-io-etcher-builds/releases/tag/v1.7.8

DaveFlashNL, 2022-03-22:
okay, better, but still. I want official build. this one won't run on m1 Macs, without using Xcode terminal cli to resign it and terminal to de-quarantine it first

thedocbwarren, 2022-03-22:
I've build for Ubuntu ARM (21.10) as well 1.7.8 in the same release tag.

thedocbwarren, 2022-03-22:
> okay, better, but still. I want official build. this one won't run on m1 Macs, without using Xcode terminal cli to resign it and terminal to de-quarantine it first

I'm working on signing this so this is not an issue.

thedocbwarren, 2022-03-27:
Signing complete.  Release 1.7.8 is now signed.

thedocbwarren, 2022-03-27:
FYI, I have v1.7.8 for Ubuntu ARM in an appimage in that release as well.

Itai-Nelken, 2022-05-19:
I have changed my repo's name to `BalenaEtcher-arm`. old link will redirect until someone else makes a repository with it (I think).
Also, [v1.7.9](https://github.com/Itai-Nelken/BalenaEtcher-arm/releases/tag/v1.7.9) is built and available for armv7l and arm64.

melyux, 2022-06-08:
@thedocbwarren How did you manage to sign your release? Did you have to pay Apple money?

thedocbwarren, 2022-06-08:
> @thedocbwarren How did you manage to sign your release? Did you have to pay Apple money?
I did sign my release.  I have an Apple Developer account for my apps so, yes, I did pay for that.

That said I'm accepting responsibility and supporting my releases as part of the signing.

melyux, 2022-06-08:
@thedocbwarren Does that mean yes?

lurch, 2022-06-08:
Looks like the quote-formatting on GitHub screwed up slightly? As part of his reply, @thedocbwarren also said "I did sign my release. I have an Apple Developer account for my apps so, yes, I did pay for that.".

melyux, 2022-06-08:
Thanks, got tripped up by the formatting. That sucks, no way to make free software without being punished for it.

thedocbwarren, 2022-06-08:
I write plenty with my account and more than happy to help with this project, so no worries!  I'm considering experimenting and seeing if I can upgrade Electron.

theofficialgman, 2022-11-17:
Etcher v1.10.0 for arm32 and arm64 Linux.
Deb rpm and appimage included. Built on Ubuntu bionic so compatible with anything glibc 2.27+ (same as official release)
https://github.com/Pi-Apps-Coders/files/releases/tag/large-files

jens-idoer, 2023-03-17:
Etcher build for Apple Silicon.

Specifically build on MacBook Pro M2 

https://bitbucket.org/idoerdk/balena-etcher-m2/src/master/1.18.5/

Was a quite straight forward build so I'm curious as to why this is not offered as a "standard" download option ?

mcraa, 2023-03-17:
> Was a quite straight forward build so I'm curious as to why this is not offered as a "standard" download option ?

The last time we tried, the mac github-hosted runners were not able to produce the correct output, most probably they are just emulated not real arm hardware, some dependencies like lzma-native had the same problem.
Current maintainers need to retry to verify it is still the case.

thedocbwarren, 2023-03-20:
@jens-idoer how were you able to build the DMG?  I'm not able to build on my end.

lockieluke, 2023-05-06:
what command did u use to build it?

jens-idoer, 2023-05-06:
Did not really make any notes .. but I remember it as being without any of the expected hoops and loops you need to go through to compile anything from opensource projects.
Next time I need to compile it I'll make notes and leave them in the readme.

thedocbwarren, 2023-05-06:
hmmmm, strange.  I was never able to build a DMG after 1.7.9.  Every build would compile and pack without any issue but never produce a DMG.

thedocbwarren, 2023-05-06:
I'm more than willing to continue to build these and support but I can't get a DMG to build.

thedocbwarren, 2023-12-29:
Published a new version!  It's in the repo for my community builds.  

version 1.19.3

https://github.com/Augmentedjs/balena-io-etcher-builds/releases/tag/v1.19.3

thedocbwarren, 2023-12-30:
🤦🏻‍♂️oh wow I missed that.  Glad to see it.