balena-io/etcher #3716: Installation fails on KDE neon (based on Ubuntu 20.04)

senarclens, 2022-03-03:
KDE Neon is based on Ubuntu 20.04 and replacing the commandline with

```
curl -1sLf    'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh' | distro=ubuntu version=20.04 codename=focal sudo -E bash
```

is a good workaround. However, you might want to add

```
... distro=neon  version=20.04  codename=focal  arch=x86_64
```

to your list of supported distros.

- **Etcher version:** current as of 2022-03-03
- **Operating system and architecture:** distro=neon version=20.04 codename=focal sudo -E bash
- **What do you think should have happened:** adding etcher to the repositories with the directions in https://github.com/balena-io/etcher?d_id=9e84538c-1fc3-4b70-a9c5-922599eacf50R#debian-and-ubuntu-based-package-repository-gnulinux-x86x64 should work
- **What happened:**

```
 NOPE: Checking if upstream install config is OK ...
 ^^^^: ... Failed to fetch configuration for your OS distribution release/version.
 >>>>:
 >>>>: It looks like we don't currently support your distribution release and
 >>>>: version. This is something that we can fix by adding it to our list of
 >>>>: supported versions (see contact us below), or you can manually override
 >>>>: the values below to an equivalent distribution that we do support:
 >>>>:
 >>>>: ... distro=neon  version=20.04  codename=focal  arch=x86_64  
 >>>>:
 >>>>: You can force this script to use a particular value by specifying distro,
 >>>>: version, or codename via environment variable. E.g., to specify a distro
 >>>>: such as Ubuntu/Xenial (16.04), use the following:
 >>>>:
 >>>>:   <curl command> | distro=ubuntu version=16.04 codename=xenial sudo -E bash
 >>>>:
 !!!!: Oh no, your setup failed! :-( ... But we might be able to help. :-)
 !!!!: 
 !!!!: You can contact balena for further assistance.
 !!!!: 
 !!!!: URL: https://github.com/balena-io/etcher
 !!!!: 
 !!!!: Last 20 log lines from .csm_setup_8OfJtcjTr (might not be errors, nor even relevant):
 !!!!:
 !!!!: > Warning: apt-key output should not be parsed (stdout is not a terminal)
 !!!!: > OK
```

xoxloviwan, 2022-03-06:
```
~/Загрузки$ curl -1sLf \
>    'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh' \
>    | sudo -E bash
Executing the  setup script for the 'balena/etcher' repository ...

   OK: Checking for required executable 'curl' ...
   OK: Checking for required executable 'apt-get' ...
   OK: Detecting your OS distribution and release using system methods ...
 ^^^^: ... Detected/provided for your OS/distribution, version and architecture:
 >>>>:
 >>>>: ... distro=neon  version=20.04  codename=focal  arch=x86_64  
 >>>>:
 NOPE: Checking for apt dependency 'apt-transport-https' ...
   OK: Updating apt repository metadata cache ...
   OK: Attempting to install 'apt-transport-https' ...
   OK: Checking for apt dependency 'ca-certificates' ...
   OK: Checking for apt dependency 'gnupg' ...
   OK: Importing 'balena/etcher' repository GPG key into apt ...
 NOPE: Checking if upstream install config is OK ...
 ^^^^: ... Failed to fetch configuration for your OS distribution release/version.
 >>>>:
 >>>>: It looks like we don't currently support your distribution release and
 >>>>: version. This is something that we can fix by adding it to our list of
 >>>>: supported versions (see contact us below), or you can manually override
 >>>>: the values below to an equivalent distribution that we do support:
 >>>>:
 >>>>: ... distro=neon  version=20.04  codename=focal  arch=x86_64  
 >>>>:
 >>>>: You can force this script to use a particular value by specifying distro,
 >>>>: version, or codename via environment variable. E.g., to specify a distro
 >>>>: such as Ubuntu/Xenial (16.04), use the following:
 >>>>:
 >>>>:   <curl command> | distro=ubuntu version=16.04 codename=xenial sudo -E bash
 >>>>:
 !!!!: Oh no, your setup failed! :-( ... But we might be able to help. :-)
 !!!!: 
 !!!!: You can contact balena for further assistance.
 !!!!: 
 !!!!: URL: https://github.com/balena-io/etcher
 !!!!: 
 !!!!: Last 20 log lines from .csm_setup_fVqXiO9Hf (might not be errors, nor even relevant):
 !!!!:
 !!!!: > Warning: apt-key output should not be parsed (stdout is not a terminal)
 !!!!: > OK

```
Same problem

mcraa, 2022-03-07:
It looks like it works exactly how our PPA provider cloudsmith has it in their [documentation](https://cloudsmith.io/~balena/repos/etcher/setup/#formats-deb)
We will forward the request to extend the supported distro list on cloudsmith, 
at the same time, you may contact them as well.

closed 2022-03-07 by mcraa