balena-io/etcher #3717: Etcher says successful but USB is blank

KTkabo0m, 2022-03-04:
- **Etcher version:**  1.7.7.0
- **Operating system and architecture:**  Windows 10 Pro
- **Image flashed:** lubuntu-16.04.6-desktop-i386.iso
- **What do you think should have happened:**  It should have Flashed the ISO file to USB Flash Drive so that it would be bootable to install on my old NetBook which the wiki said it would be compatible for. 
- **What happened:** Etcher said Flash complete and 1 Successful target and Effective speed: 5.2MB/s but the target which was the USB Flash Stick drive is empty and no longer reads and Windows says it has to be formatted. Why is nothing on there if the Flash was successful? I am confused.
- **Do you see any meaningful error information in the DevTools?** I don't have the DevTools. Will that fix it or tell me what is wrong? Where do I get these DevTools?
[image]
[image]
[image]
[image]
[image]
[image]

mcraa, 2022-03-04:
Of course you have dev tools @KTkabo0m It is built into the browser. Just read the issue template carefully.
[image]

Linux partitions are not necessarily visible on windows.
Did you try to boot from the drive or just assumed something is wrong because windows is not showing it?

lurch, 2022-03-04:
@mcraa 

> Linux partitions are not necessarily visible on windows.

That seems to confuse people on a regular basis - it might be worth adding something to https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md ?

KTkabo0m, 2022-03-05:
> Of course you have dev tools @KTkabo0m It is built into the browser. Just read the issue template carefully. [image]
> 
> Linux partitions are not necessarily visible on windows. Did you try to boot from the drive or just assumed something is wrong because windows is not showing it?

Well I stopped using Etcher after trying it 4 times. I then switched to Rufus which worked on the first try. The same ISO and I could see that the USB flash drive had content when I used Rufus but nothing was on it with Etcher. 

So now that I have the work done using Rufus I guess this can now be closed. 

Thank you.

closed 2022-03-05 by mcraa

lurch, 2022-03-05:
Rufus works in a slightly different manner than Etcher does #413
(which means that Rufus only works with disk images that it explicitly supports, whereas Etcher works with _any_ disk image)