balena-io/etcher #3720: - etcher 1.7.7

m4x15, 2022-03-06:
- etcher 1.7.7
- **windows 11 operating system
- **VOLUMIO:** 
- **What do you think should have happened:** it should have actually flashed 
- **What happened:* i have tried many many many times to flash volumio onto a 8gb and a 32gb micro sd for my raspberry pi as an audio player no matter what i do it will not flash i have tried re installing i have tried restating,looking up if anyone else has has this problem i have been trying for 4 days now and nothing is working. it gets to 100% on decompressing then says it has failed it said it worked one time after it failed 100 times but it didn't actually flash onto the sd I'm not sure what to do I have no other info that's it

mcraa, 2022-03-07:
Did you try to boot from the drive after the first (successful) flash?
Linux partitions most of the time are not visible in windows.
You may need to [reformat](https://www.balena.io/blog/did-etcher-break-my-usb-sd-card/) your drive to try again

m4x15, 2022-03-07:
Yes I did I put it into the pi and let it boot for a couple mins then I went to it's up and the local webpage for volumio and it didn't work

closed 2022-04-06 by m4x15