balena-io/etcher #3722: White screen in windows 10 32bit

el3rg, 2022-03-08:
- **Etcher version:** 1.7.7
- **Operating system and architecture:** windows 10 32bit
- **Image flashed:** no 
- **Do you see any meaningful error information in the DevTools?**  ERR_FILE_NOT_FOUND

-When i open the program both prortable & installed it gives a white screen. 
-When i open devtool it gives me to errors 2.
-The first the problem doesn't found index.css (the file isn't in the directory).
-The second the problem doesn't found binding.node (the file is in the directory) it doesn't recognize the file.

[image]

[image]

hooddy, 2022-11-08:
have the same issue in Win7x64 start from 1.7.10 till latest 1.8.x. version 1.7.9 looks fine.

Anton-V-K, 2023-11-27:
The mentioned module `C:\Users\???\AppData\Local\Temp\*****\resources\app\generated\modules\@ronomon\direct-io\binding.node` is a 64-bit binary even when deployed by 1.7.9 portable version, which is supposed to run on 32-bit Windows.
Moreover other deployed modules like (`...\resources\app\generated\modules\@balena.io\usb\build\Release\usb_bindings.node`, `...\resources\app\generated\modules\drivelist\build\Release\drivelist.node`, etc.) are also 64-bit.
So if you want to run the app on 32-bit Windows, you should use an older version.