balena-io/etcher #3724: Fake etcher sites

mcraa, 2022-03-10:
Please read and stay alert:
https://www.balena.io/blog/beware-false-software-claiming-to-be-balenaetcher/

In case you find similar malicious sites you may paste it here as a text like `etcheer.com` or screenshot, not to help them with more links pointing to them.

lurch, 2022-03-11:
Yikes! I guess that's the price for being so popular :cry: 

With my eagle-eyes I spotted that the first paragraph says "only download balenaEtcher from our official sources: the [balena.io website](https://www.balena.io/etcher/), or [GitHub repository](https://github.com/balena-io/etcher/releases)." but a later paragraph says "If you click the download link on the http://etcher.io website, or go to https://github.com/balena-io/etcher" :eyes: 

I know that `http://etcher.io` just redirects to `https://www.balena.io/etcher/`, but it seems a little contradictory to have them both mentioned in the blog article.

lurch, 2022-03-11:
`https://balena-etcher.com/` is claiming to be an "official website" :slightly_frowning_face:

andrewnhem, 2022-03-11:
Thanks for the feedback here @lurch . Refined the post a little to reflect your recommendation. Feedback is always welcome.

lurch, 2022-03-11:
...and there's also `https://balena-etcher.eu/` (which is slightly different to the one that popped up in #3650 )

[image]

mcraa, 2022-03-14:
@lurch the site in the blog post is not always showing an etcher site copy, sometimes it displays some kind of ad for an engraving tool. Probably trying to be less discoverable while still spreading their harm. But the effort put into this confirms that want to be harmful on purpose.
Long story short, a suspicious 'etcher' containing URL, even if it is not showing an etcher site at the moment, should be considered a scam.

davidak, 2022-05-27:
https://etcher.download/
https://www.etcher.net/
https://etcherpc.com/
https://appimage.github.io/Etcher/ (this looks kind of official, but outdated)

and all those nasty download sites

rradar, 2022-08-14:
What I don't understand why do these people fake the sites so badly? It would be so easy just to copy the "original" site (would probably also save some a lot of time instead of creating a "new" one). Something that would be really funny if someone pushes a "fake" site with a etcher version that doesn't include the telemetries, adware etc. the original etcher ships :rofl:

davidak, 2022-08-16:
@rradar that seem to be "good enough" to scam people. and the scammers might not have the technical knowledge to download a website (i guess even the browser can just save it lol)

there are alternatives to etcher that don't have telemetry and are not based on electron, like [usbimager](https://bztsrc.gitlab.io/usbimager/), but they are not that userfriendly. i suggest both and try to help usbimager to become more user-friendly, so etcher can be abandoned

rradar, 2022-08-21:
> not that userfriendly

How's that? Without "telemetry", ads and other "[unintentional](https://github.com/balena-io/etcher/issues/2766#issuecomment-531437542)" connections to the biggest data driven companies on planet earth a program can't be userfriendly? Or is that a project with [less than 1](https://gitlab.com/bztsrc/usbimager/-/issues)00 (or [300](https://github.com/balena-io/etcher/issues)) open issues can't be userfriendly? Or might be the program size - anything which isn't hundreds of megabytes worth of download can't be userfriendly?

How simpler can a program be where the user needs to choose exactly two things (source & target)? :thinking: 
>[image]

lurch, 2022-08-22:
For some people, a button labelled `...` is probably less userfriendly than a button labelled `Select image` ? :shrug:

davidak, 2022-08-25:
> Without "telemetry", ads and other [...] a program can't be userfriendly?

This is not the place to discuss UX. I created an issue here: https://gitlab.com/bztsrc/usbimager/-/issues/87

rradar, 2022-09-01:
I'm always baffled **when users or creators think their own user experience is valid for all people** on earth :earth_asia:

Did one (@lurch, @davidak ..) ever gave a thought that **people maybe not blessed with the horse power your computers can deliver**? I remember when I tried to open etcher many years ago virtually nothing showed - just my system start lagging. It took very long time till the gui appeared. What do you think? Could people eventually like to trade that colored user interface with icons and animation in favor of a fast, responsive program or not?

Also people from Germany or Europe often easily tend to forget that not the whole world has **access to fast or even unmetered internet**! In many countries there simply is no flat rates available - every byte counts/costs! Again the question (if you could put yourself in such a position) what would your experience look like with etcher? A 300mb download and roughly another 300mb dependencies - maybe 3 hours download time. Or could would your (overall) experience maybe greatly improve if there would be a program that you could download in less than 1 minute (and that doesn't cost you like $5 in traffic) and virtually does the same - or maybe even better?

lurch, 2022-09-01:
If you personally don't want to use Etcher, no-one is forcing you to? :shrug:

ldo, 2022-11-12:
If you published SHA-256 hashes for the downloads, then people needn’t worry where they got them from, just so long as the hashes match.

lurch, 2022-11-13:
> If you published SHA-256 hashes for the downloads

See https://github.com/balena-io/etcher/pull/3839

> then people needn’t worry where they got them from, just so long as the hashes match.

How many people do you know who actually bother verifying the checksums of all programs they download, before actually running them?
Also, the "dodgy" Etcher websites could still publish the SHA-256 hashes of the "dodgy" Etcher downloads they're offering? :thinking:

ldo, 2022-11-13:
You could include the correct ones in the announcements to the usual channels.

oghaki, 2023-06-13:
Not sure if this has already been accounted for, but <balenaetcher.org> popped up when I searched today.