balena-io/etcher #3726: My USB Pen Drive disappears after flashing from Balena Etcher

mahir28, 2022-03-11:
- **Etcher version:**  1.7.7
- **Operating system and architecture:**   Windows 10 21H1
- **Image flashed:**  HiveOS
- **What happened:** I flashed a ``.img.xz`` file downloaded from HiveOS website and flash in my pen drive. After successfully flashing and validating my pen drive just disappeared from everywhere. I did not find it anywhere in file explorer or device manager. But it shows in Boot Menu. But it also does not boot properly in Boot Menu also. Now how can I make my Pen Drive visible again?
- **Do you see any meaningful error information in the DevTools? :** Nope No error in Dev Tools

mcraa, 2022-03-11:
does this help? https://www.balena.io/blog/did-etcher-break-my-usb-sd-card/

mahir28, 2022-03-11:
> does this help? https://www.balena.io/blog/did-etcher-break-my-usb-sd-card/

I didn't even see my USB in DiskPart

Nmatt1, 2022-03-18:
HI there @mahir28 hope you are doing well.

When I flashed HiveOS using Balena Etcher a few weeks ago I had no issues with this.

Your USB stick will disappear after having an operating system written to the USB, Windows wont be able to read the contents correctly.

To boot the HiveOS on the USB, you need to do the following things.

- [ ] Turn off computer
- [ ] Make sure USB stick with HiveOS is plugged in
- [ ] Turn computer on
- [ ] Enter your motherboards BIOS using the BIOS key (check your motherboard manual for which key it is)
- [ ] Select the USB in boot order and restart
- [ ] You will be booted in HiveOS

From there you are good.

If you want to clear the data off the USB stick and have it usable again do the following things

- [ ] Download the [GParted](https://gparted.org/download.php) ISO
- [ ] Get another USB stick
- [ ] Use Balena Etcher to write the GParted ISO to the 2nd USB stick
- [ ] Following the boot instructions from above, boot to the GParted USB stick
- [ ] Then using [this](https://www.youtube.com/watch?v=gwFzVc20qTk) tutorial to wipe your HiveOS USB stick and you are safe to turn off PC and go back to your operating system.

I hope you were able to help and I hope you have a good rest of your day!