balena-io/etcher #3736: Flashing fails for some SDs but not others

aalami90, 2022-04-01:
- **Etcher version:** 1.7.8.0
- **Operating system and architecture:** Windows 10 Enterprise 20H2, 64-bit operating system, x64-based processor 
- **Image flashed:** otbr.img (Open Thread Border Router, for Raspberry PI, specifically the one for OTS_BR)
- **What do you think should have happened: ** Flashing should pass
- **What happened: ** Flashing fails in validation stage
- **Do you see any meaningful error information in the DevTools?** Here is the log

Flashing 1 device, 0% at 14.20 MB/s (total 14.20 MB/s) eta in 271s with 0 failed devices

 Flashing 1 device, 0% at 11.80 MB/s (total 11.80 MB/s) eta in 326s with 0 failed devices
 Flashing 1 device, 0% at 10.80 MB/s (total 10.80 MB/s) eta in 356s with 0 failed devices
 Flashing 1 device, 1% at 10.70 MB/s (total 10.70 MB/s) eta in 356s with 0 failed devices
 Flashing 1 device, 1% at 11.20 MB/s (total 11.20 MB/s) eta in 341s with 0 failed devices
 Flashing 1 device, 1% at 10.60 MB/s (total 10.60 MB/s) eta in 357s with 0 failed devices
2 Flashing 1 device, 2% at 10.70 MB/s (total 10.70 MB/s) eta in 352s with 0 failed devices
 Flashing 1 device, 2% at 10.70 MB/s (total 10.70 MB/s) eta in 353s with 0 failed devices
 Flashing 1 device, 2% at 11.10 MB/s (total 11.10 MB/s) eta in 338s with 0 failed devices
 Flashing 1 device, 3% at 11.00 MB/s (total 11.00 MB/s) eta in 339s with 0 failed devices
2 Flashing 1 device, 3% at 10.60 MB/s (total 10.60 MB/s) eta in 351s with 0 failed devices
2 Flashing 1 device, 4% at 11.10 MB/s (total 11.10 MB/s) eta in 334s with 0 failed devices
 Flashing 1 device, 4% at 11.00 MB/s (total 11.00 MB/s) eta in 334s with 0 failed devices
 Flashing 1 device, 4% at 10.70 MB/s (total 10.70 MB/s) eta in 343s with 0 failed devices
 Flashing 1 device, 5% at 10.70 MB/s (total 10.70 MB/s) eta in 341s with 0 failed devices
 Flashing 1 device, 5% at 10.70 MB/s (total 10.70 MB/s) eta in 340s with 0 failed devices
 Flashing 1 device, 5% at 10.70 MB/s (total 10.70 MB/s) eta in 339s with 0 failed devices
 Flashing 1 device, 6% at 11.10 MB/s (total 11.10 MB/s) eta in 327s with 0 failed devices
 Flashing 1 device, 6% at 11.00 MB/s (total 11.00 MB/s) eta in 328s with 0 failed devices
 Flashing 1 device, 6% at 10.60 MB/s (total 10.60 MB/s) eta in 339s with 0 failed devices
 Flashing 1 device, 7% at 10.60 MB/s (total 10.60 MB/s) eta in 340s with 0 failed devices
 Flashing 1 device, 7% at 11.00 MB/s (total 11.00 MB/s) eta in 325s with 0 failed devices
 Flashing 1 device, 7% at 11.20 MB/s (total 11.20 MB/s) eta in 318s with 0 failed devices
 Flashing 1 device, 7% at 10.70 MB/s (total 10.70 MB/s) eta in 334s with 0 failed devices
 Flashing 1 device, 8% at 10.50 MB/s (total 10.50 MB/s) eta in 337s with 0 failed devices
 Flashing 1 device, 8% at 11.00 MB/s (total 11.00 MB/s) eta in 320s with 0 failed devices
 Flashing 1 device, 8% at 11.00 MB/s (total 11.00 MB/s) eta in 319s with 0 failed devices
 Flashing 1 device, 9% at 11.10 MB/s (total 11.10 MB/s) eta in 316s with 0 failed devices
 Flashing 1 device, 9% at 10.70 MB/s (total 10.70 MB/s) eta in 327s with 0 failed devices
 Flashing 1 device, 9% at 10.70 MB/s (total 10.70 MB/s) eta in 326s with 0 failed devices
 Flashing 1 device, 9% at 10.60 MB/s (total 10.60 MB/s) eta in 328s with 0 failed devices
 Flashing 1 device, 10% at 10.60 MB/s (total 10.60 MB/s) eta in 326s with 0 failed devices
 Flashing 1 device, 10% at 11.20 MB/s (total 11.20 MB/s) eta in 310s with 0 failed devices
 Flashing 1 device, 10% at 10.60 MB/s (total 10.60 MB/s) eta in 324s with 0 failed devices
 Flashing 1 device, 11% at 10.60 MB/s (total 10.60 MB/s) eta in 324s with 0 failed devices
 Flashing 1 device, 11% at 11.20 MB/s (total 11.20 MB/s) eta in 307s with 0 failed devices
 Flashing 1 device, 11% at 10.70 MB/s (total 10.70 MB/s) eta in 319s with 0 failed devices
 Flashing 1 device, 11% at 10.80 MB/s (total 10.80 MB/s) eta in 315s with 0 failed devices
 Flashing 1 device, 12% at 11.20 MB/s (total 11.20 MB/s) eta in 302s with 0 failed devices
 Flashing 1 device, 12% at 10.60 MB/s (total 10.60 MB/s) eta in 318s with 0 failed devices
 Flashing 1 device, 12% at 10.60 MB/s (total 10.60 MB/s) eta in 317s with 0 failed devices
 Flashing 1 device, 13% at 11.10 MB/s (total 11.10 MB/s) eta in 303s with 0 failed devices
 Flashing 1 device, 13% at 10.60 MB/s (total 10.60 MB/s) eta in 317s with 0 failed devices
 Flashing 1 device, 13% at 10.60 MB/s (total 10.60 MB/s) eta in 313s with 0 failed devices
 Flashing 1 device, 13% at 11.20 MB/s (total 11.20 MB/s) eta in 298s with 0 failed devices
 Flashing 1 device, 14% at 10.70 MB/s (total 10.70 MB/s) eta in 311s with 0 failed devices
 Flashing 1 device, 14% at 10.70 MB/s (total 10.70 MB/s) eta in 308s with 0 failed devices
 Flashing 1 device, 14% at 11.10 MB/s (total 11.10 MB/s) eta in 296s with 0 failed devices
 Flashing 1 device, 15% at 10.50 MB/s (total 10.50 MB/s) eta in 311s with 0 failed devices
 Flashing 1 device, 15% at 10.60 MB/s (total 10.60 MB/s) eta in 308s with 0 failed devices
 Flashing 1 device, 15% at 10.90 MB/s (total 10.90 MB/s) eta in 297s with 0 failed devices
 Flashing 1 device, 16% at 11.00 MB/s (total 11.00 MB/s) eta in 296s with 0 failed devices
 Flashing 1 device, 16% at 11.10 MB/s (total 11.10 MB/s) eta in 292s with 0 failed devices
 Flashing 1 device, 16% at 11.10 MB/s (total 11.10 MB/s) eta in 291s with 0 failed devices
 Flashing 1 device, 16% at 10.60 MB/s (total 10.60 MB/s) eta in 302s with 0 failed devices
 Flashing 1 device, 17% at 10.60 MB/s (total 10.60 MB/s) eta in 300s with 0 failed devices
 Flashing 1 device, 17% at 11.10 MB/s (total 11.10 MB/s) eta in 286s with 0 failed devices
 Flashing 1 device, 17% at 10.60 MB/s (total 10.60 MB/s) eta in 299s with 0 failed devices
 Flashing 1 device, 18% at 10.60 MB/s (total 10.60 MB/s) eta in 300s with 0 failed devices
 Flashing 1 device, 18% at 10.60 MB/s (total 10.60 MB/s) eta in 299s with 0 failed devices
 Flashing 1 device, 18% at 10.60 MB/s (total 10.60 MB/s) eta in 298s with 0 failed devices
 Flashing 1 device, 18% at 11.10 MB/s (total 11.10 MB/s) eta in 281s with 0 failed devices
[... 298 lines cut ...]
 Verifying 1 device, 37% at 71.50 MB/s (total 71.50 MB/s) eta in 34s with 0 failed devices
 Verifying 1 device, 39% at 70.80 MB/s (total 70.80 MB/s) eta in 33s with 0 failed devices
 Verifying 1 device, 41% at 70.00 MB/s (total 70.00 MB/s) eta in 32s with 0 failed devices
 Verifying 1 device, 43% at 70.00 MB/s (total 70.00 MB/s) eta in 31s with 0 failed devices
 Verifying 1 device, 45% at 69.90 MB/s (total 69.90 MB/s) eta in 30s with 0 failed devices
 Verifying 1 device, 47% at 70.00 MB/s (total 70.00 MB/s) eta in 29s with 0 failed devices
 Verifying 1 device, 48% at 69.80 MB/s (total 69.80 MB/s) eta in 28s with 0 failed devices
 Verifying 1 device, 50% at 70.50 MB/s (total 70.50 MB/s) eta in 27s with 0 failed devices
 Verifying 1 device, 52% at 70.40 MB/s (total 70.40 MB/s) eta in 26s with 0 failed devices
 Verifying 1 device, 54% at 70.80 MB/s (total 70.80 MB/s) eta in 25s with 0 failed devices
 Verifying 1 device, 56% at 70.40 MB/s (total 70.40 MB/s) eta in 24s with 0 failed devices
 Verifying 1 device, 58% at 68.40 MB/s (total 68.40 MB/s) eta in 24s with 0 failed devices
 Verifying 1 device, 59% at 67.60 MB/s (total 67.60 MB/s) eta in 23s with 0 failed devices
 Verifying 1 device, 61% at 66.70 MB/s (total 66.70 MB/s) eta in 22s with 0 failed devices

gui.js:46 Fri Apr 01 2022 18:04:10 GMT+0200 (Central European Summer Time) Validation error ({"image":{"size":4024762880,"name":"otbr.img","hasMBR":true,"partitions":[{"offset":4194304,"size":268435456,"type":12,"index":1},{"offset":272629760,"size":3752133120,"type":131,"index":2}],"extension":"img","path":"C:\\Users\\nxf85341\\Desktop\\otbr.img"},"drives":[{"size":15502147584,"isVirtual":false,"enumerator":"PCISTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"D:\\, "}],"description":"SDHC Card","isUAS":false,"partitionTableType":"mbr"}],"driveCount":1,"uuid":"4505b3c7-621a-47c8-ad00-722af7fda922","flashInstanceUuid":"4505b3c7-621a-47c8-ad00-722af7fda922","sample":0.1,"applicationSessionUuid":"13df90f2-ca06-406e-a533-029f7bbc26f1","flashingWorkflowUuid":"b2e1a4fa-e0a7-4be6-9148-5d79260f2f62"})
gui.js:35 Verifying 1 device, 63% at 66.20 MB/s (total 66.20 MB/s) eta in 22s with 0 failed devices
gui.js:46 Finish: 2435842048
gui.js:46 Terminating IPC server
gui.js:46 Flash results {results: {…}, cancelled: false, skip: false}
[... cut ...]
gui.js:46 Fri Apr 01 2022 18:06:54 GMT+0200 (Central European Summer Time) Restart ({"sample":0.1,"applicationSessionUuid":"13df90f2-ca06-406e-a533-029f7bbc26f1","flashingWorkflowUuid":"b2e1a4fa-e0a7-4be6-9148-5d79260f2f62"})
gui.js:46 Fri Apr 01 2022 18:06:55 GMT+0200 (Central European Summer Time) Open external link ({"url":"https://github.com/balena-io/etcher/blob/master/SUPPORT.md","sample":0.1,"applicationSessionUuid":"13df90f2-ca06-406e-a533-029f7bbc26f1","flashingWorkflowUuid":"51e85cc9-498f-4d65-b582-165abed64c6a"})

aalami90, 2022-04-01:
gui.js:46 Flash results {results: {…}, cancelled: false, skip: false}cancelled: falseresults: averageFlashingSpeed: 10630760.570821617bytesWritten: 2435842048devices: failed: 1successful: 0__proto__: constructor: ƒ Object()hasOwnProperty: ƒ hasOwnProperty()isPrototypeOf: ƒ isPrototypeOf()arguments: (...)caller: (...)length: 1name: "isPrototypeOf"__proto__: ƒ ()[[Scopes]]: Scopes[0]propertyIsEnumerable: ƒ propertyIsEnumerable()toLocaleString: ƒ toLocaleString()toString: ƒ toString()valueOf: ƒ valueOf()__defineGetter__: ƒ __defineGetter__()__defineSetter__: ƒ __defineSetter__()__lookupGetter__: ƒ __lookupGetter__()__lookupSetter__: ƒ __lookupSetter__()get __proto__: ƒ __proto__()set __proto__: ƒ __proto__()errors: Array(1)0: Error: Checksum does not match for range [0, 2446327807]: "��1�~�#�" != "�^��30��"
    at verifyOrGenerateChecksum (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:165220)
    at createSparseReaderStateIterator (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:164787)
    at createSparseReaderStateIterator.next (<anonymous>)
    at SparseReadStream.nextBlock (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:166398)
    at SparseReadStream.__read (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:167206)
    at async SparseReadStream._read (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:166280)code: "EVALIDATION"description: "SDHC Card"device: "\\.\PhysicalDrive1"name: "Error"message: "Checksum does not match for range [0, 2446327807]: "��1�~�#�" != "�^��30��""stack: "Error: Checksum does not match for range [0, 2446327807]: "��1�~�#�" != "�^��30��"↵    at verifyOrGenerateChecksum (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:165220)↵    at createSparseReaderStateIterator (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:164787)↵    at createSparseReaderStateIterator.next (<anonymous>)↵    at SparseReadStream.nextBlock (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:166398)↵    at SparseReadStream.__read (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:167206)↵    at async SparseReadStream._read (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:166280)"__proto__: constructor: ƒ Error()message: ""name: "Error"toString: ƒ toString()__proto__: Objectlength: 1__proto__: Array(0)concat: ƒ concat()constructor: ƒ Array()copyWithin: ƒ copyWithin()entries: ƒ entries()every: ƒ every()fill: ƒ fill()filter: ƒ filter()find: ƒ find()findIndex: ƒ findIndex()flat: ƒ flat()flatMap: ƒ flatMap()forEach: ƒ forEach()includes: ƒ includes()indexOf: ƒ indexOf()join: ƒ join()keys: ƒ keys()lastIndexOf: ƒ lastIndexOf()length: 0map: ƒ map()pop: ƒ pop()push: ƒ push()reduce: ƒ reduce()reduceRight: ƒ reduceRight()reverse: ƒ reverse()shift: ƒ shift()slice: ƒ slice()some: ƒ some()sort: ƒ sort()splice: ƒ splice()toLocaleString: ƒ toLocaleString()toString: ƒ toString()unshift: ƒ unshift()values: ƒ values()Symbol(Symbol.iterator): ƒ values()Symbol(Symbol.unscopables): {copyWithin: true, entries: true, fill: true, find: true, findIndex: true, …}__proto__: ObjectsourceMetadata: {size: 4024762880, name: "otbr.img", blocks: Array(9), blockmappedSize: 3863282176}__proto__: Objectskip: false__proto__: constructor: ƒ Object()hasOwnProperty: ƒ hasOwnProperty()isPrototypeOf: ƒ isPrototypeOf()propertyIsEnumerable: ƒ propertyIsEnumerable()toLocaleString: ƒ toLocaleString()toString: ƒ toString()valueOf: ƒ valueOf()__defineGetter__: ƒ __defineGetter__()__defineSetter__: ƒ __defineSetter__()__lookupGetter__: ƒ __lookupGetter__()__lookupSetter__: ƒ __lookupSetter__()get __proto__: ƒ __proto__()set __proto__: ƒ __proto__()
gui.js:46 Fri Apr 01 2022 18:04:13 GMT+0200 (Central European Summer Time) Done ({"image":{"size":4024762880,"name":"otbr.img","hasMBR":true,"partitions":[{"offset":4194304,"size":268435456,"type":12,"index":1},{"offset":272629760,"size":3752133120,"type":131,"index":2}],"extension":"img","path":"C:\\Users\\nxf85341\\Desktop\\otbr.img"},"drives":[{"size":15502147584,"isVirtual":false,"enumerator":"PCISTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"D:\\, "}],"description":"SDHC Card","isUAS":false,"partitionTableType":"mbr"}],"driveCount":1,"uuid":"4505b3c7-621a-47c8-ad00-722af7fda922","status":"finished","flashInstanceUuid":"4505b3c7-621a-47c8-ad00-722af7fda922","errors":[{"name":"Error","description":"SDHC Card","code":"EVALIDATION","device":"\\\\.\\PhysicalDrive1"}],"devices":{"failed":1,"successful":0},"bytesWritten":2435842048,"sourceMetadata":{"size":4024762880,"name":"otbr.img","blocks":[{"blocks":[{"offset":0,"length":2446327808}]},{"blocks":[{"offset":2453667840,"length":1267728384}]},{"blocks":[{"offset":3737124864,"length":24117248}]},{"blocks":[{"offset":3765436416,"length":1048576}]},{"blocks":[{"offset":3770679296,"length":8388608}]},{"blocks":[{"offset":3896508416,"length":4194304}]},{"blocks":[{"offset":3904897024,"length":34603008}]},{"blocks":[{"offset":3946840064,"length":32505856}]},{"blocks":[{"offset":3980394496,"length":44368384}]}],"blockmappedSize":386 [... cut ...]
gui.js:46 Fri Apr 01 2022 18:06:54 GMT+0200 (Central European Summer Time) Restart ({"sample":0.1,"applicationSessionUuid":"13df90f2-ca06-406e-a533-029f7bbc26f1","flashingWorkflowUuid":"b2e1a4fa-e0a7-4be6-9148-5d79260f2f62"})
gui.js:46 Fri Apr 01 2022 18:06:55 GMT+0200 (Central European Summer Time) Open external link ({"url":"https://github.com/balena-io/etcher/blob/master/SUPPORT.md","sample":0.1,"applicationSessionUuid":"13df90f2-ca06-406e-a533-029f7bbc26f1","flashingWorkflowUuid":"51e85cc9-498f-4d65-b582-165abed64c6a"})

aalami90, 2022-04-01:
Flash results 
{results: {…}, cancelled: false, skip: false}
cancelled: false
results:
averageFlashingSpeed: 10630760.570821617
bytesWritten: 2435842048
devices:
failed: 1
successful: 0
__proto__: Object
errors: Array(1)
0: Error: Checksum does not match for range [0, 2446327807]: "��1�~�#�" != "�^��30��" at verifyOrGenerateChecksum (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:165220) at createSparseReaderStateIterator (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:164787) at createSparseReaderStateIterator.next (<anonymous>) at SparseReadStream.nextBlock (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:166398) at SparseReadStream.__read (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:167206) at async SparseReadStream._read (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:166280)
code: "EVALIDATION"
description: "SDHC Card"
device: "\\.\PhysicalDrive1"
name: "Error"
message: "Checksum does not match for range [0, 2446327807]: "��1�~�#�" != "�^��30��""
stack: "Error: Checksum does not match for range [0, 2446327807]: "��1�~�#�" != "�^��30��"↵    at verifyOrGenerateChecksum (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:165220)↵    at createSparseReaderStateIterator (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:164787)↵    at createSparseReaderStateIterator.next (<anonymous>)↵    at SparseReadStream.nextBlock (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:166398)↵    at SparseReadStream.__read (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:167206)↵    at async SparseReadStream._read (C:\Users\nxf85341\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js:1:166280)"
__proto__: Object
length: 1
__proto__: Array(0)
sourceMetadata:
blockmappedSize: 3863282176
blocks: (9) [{…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}, {…}]
name: "otbr.img"
size: 4024762880
__proto__: Object
__proto__: Object
skip: false
__proto__:
constructor: ƒ Object()
hasOwnProperty: ƒ hasOwnProperty()
isPrototypeOf: ƒ isPrototypeOf()
propertyIsEnumerable: ƒ propertyIsEnumerable()
toLocaleString: ƒ toLocaleString()
toString: ƒ toString()
valueOf: ƒ valueOf()
__defineGetter__: ƒ __defineGetter__()
__defineSetter__: ƒ __defineSetter__()
__lookupGetter__: ƒ __lookupGetter__()
__lookupSetter__: ƒ __lookupSetter__()
get __proto__: ƒ __proto__()
set __proto__: ƒ __proto__()

mcraa, 2022-04-13:
Based on the info added, I cannot find the image flashed, on the OT website there are instructions to build and install, then links only to the original raspberry OS images.
The error message says `Checksum does not match for range ...`  can be caused by a corrupted file or destination.

closed 2023-07-12 by dfunckt