balena-io/etcher #3743: Checking for update Update for version 1.7.8 is not available (latest version: 1.7.8, downgrade is disallowed).

Firestar-Reimu, 2022-04-13:
- **Etcher version: 1.7,8** 
- **Operating system and architecture: Manjaro Linux** 

When I running balena-etcher, it continuously say: 

`Checking for update
Update for version 1.7.8 is not available (latest version: 1.7.8, downgrade is disallowed).`

mcraa, 2022-04-13:
And what it says is correct. 
What is your actual issue?

Firestar-Reimu, 2022-04-14:
As this is the latest version, “Update for version 1.7.8 is not available” is not necessary.

mcraa, 2022-05-26:
yes @thedocbwarren .9 was meaningful for debian based distros and the notification was not set to be sent to the users.
See stagingpercentage in [latest*.yml](https://github.com/balena-io/etcher/releases/tag/v1.7.9) files.

However this issue was opened before .9 was out and at the time .8 was the latest version.