balena-io/etcher #3744: Feature Request: Flash Individual Partitions

zwhitchcox, 2022-04-18:
This is a common request on support.

Etcher probably will never flash system drives, but it could flash individual partitions on an external drive.

lurch, 2022-04-19:
I'm just guessing, but I suspect people don't want to **just** flash individual partitions, but are _actually_ wanting multiboot support? :man_shrugging:  See #414 and the various other issues it links to.

zwhitchcox, 2022-04-19:
Closing as a duplicate of #414 ...thanks for the heads up @lurch

closed 2022-04-19 by zwhitchcox

GrayedFox, 2022-08-23:
> I'm just guessing, but I suspect people don't want to **just** flash individual partitions, but are _actually_ wanting multiboot support? 🤷‍♂️ See #414 and the various other issues it links to.

This is a good observation but I would argue that it makes sense to keep this issue open and treat it as a stand alone topic, albeit one that could be used as a stepping stone towards multiboot in the future?

My use case is having a bootable Linux distro on one partition and simple FAT32 storage on the other. This isn't technically multiboot, I just want to be able to select an individual partition on my portable USB instead of selecting the entire drive.

What's more is it seems like the [Drive interface](https://github.com/balena-io-modules/drivelist/blob/master/lib/index.ts#L31) already supports this - that is - no changes to the Drive module would be needed? When the Etcher GUI renders the drive options it also [displays different drive partitions](https://github.com/balena-io/etcher/blob/master/lib/gui/app/components/drive-selector/drive-selector.tsx#L229) as different locations.

As far as I can tell, the main work that would need to be done to allow Etcher users to select individual partitions to flash instead of the entire drive is a reworking of the [available drives](https://github.com/balena-io/etcher/blob/master/lib/gui/app/models/available-drives.ts) and [drive constraints](https://github.com/balena-io/etcher/blob/master/lib/shared/drive-constraints.ts) model/files and then updating the GUI to make it nice -- maybe adding an List Partitions As Individual Drives link to the drive selector UI, underneath the existing Show Hidden Drives link?

[image]

GrayedFox, 2022-08-23:
Also for anyone looking to easily flash an ISO to a partition instead of being forced to use the whole drive give [http://unetbootin.github.io/](http://unetbootin.github.io/) a look :)

zwhitchcox, 2022-08-23:
The problem is that you need a [bootloader](https://en.wikipedia.org/wiki/Booting#Second-stage_boot_loader) to load the kernel, so simply flashing to a partition is not going to achieve the desired effect. The operating system also has its own partitions, which the MBR/GPT would not know about, because the partition tables are located *within* another partition, so even if you did somehow load the kernel, the partition table would not be what the OS expected.

This is what I assume netbootin does. It adds a boot loader, extracts the partitions from the OS and loads them into its own partition table and then transfers control to a volume boot record based on the selection you made.

I agree that all this would be extremely cool if Etcher could do it, but unfortunately we are not ready to take on a project like this at this time.

Right now we are focused on the tooling that will enable projects like this in the future and making it easier for developers to contribute to Etcher and add solutions like this, although I look forward to when we can implement something like this in the future!

zwhitchcox, 2022-08-23:
To be honest, this is not too terribly difficult, as a lot of the functionality is already there in other tools like GRUB, but it's better to create the tooling we are focused on before we add complexity like this, as it will create an opportunity for anyone to add functionality like this and extend Etcher.

GrayedFox, 2022-08-24:
I understand and your priorities make sense :) I look forward to seeing some plugin functionality added and then I might go ahead and tackle this myself, been meaning to get familiar with Electron for a while just to see what all the fuss is about!

Keep up the good work 🦖

ibay770, 2022-11-27:
> I'm just guessing, but I suspect people don't want to **just** flash individual partitions, but are _actually_ wanting multiboot support? 🤷‍♂️ See #414 and the various other issues it links to.

You're right, but I want to multiboot various versions of Mac OS, and the only way to do it at the moment, is unpack every installer to a different partition on the drive. They don't have a bootable iso setup like windows/linux does as yet.