balena-io/etcher #3745: Support burning Windows 11 images

spencerthayer, 2022-04-20:
I would like to request that Etcher again works on figuring out the bootable Windows ISO problem on Linux or Mac. This isn't a new issue and there are other issues like [#210](https://github.com/balena-io/etcher/issues/210) that make the same request. But they all seem to be closed without resolving the issue.

- **Etcher version:** 
Etcher Version 1.7.7 (1.7.7)
- **Operating system and architecture:** 
MacBook Pro (16-inch, 2019)
2.3 GHz 8-Core Intel Core i9
16 GB 2667 MHz DDR4
- **Image flashed:** 
Win11_English_x64v1.iso

Neolo, 2022-07-30:
Windows 11 not getting flashed, no partitions found error. So stupid...

lurch, 2022-07-31:
https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable

Neolo, 2022-07-31:
> https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable

Special treatment? Then what is the point of Balena at all??? Make that treatment. How dumb!

lurch, 2022-07-31:
> what is the point of Balena at all???

https://www.balena.io/what-is-balena

Neolo, 2022-07-31:
> > what is the point of Balena at all???
> 
> https://www.balena.io/what-is-balena

Useless junk? dd in terminal does the same.

spencerthayer, 2022-07-31:
> dd in terminal does the same.

Yup.