balena-io/etcher #3749: balenaEtcher unable to start

7217043955, 2022-04-24:
- **Etcher version:** 
- balenaEtcher-Portable-1.7.9.exe
- **Operating system and architecture:** 
- Windows 10 Pro for Workstation
- Version 21H2 (OS Build 19044.1645)
- **Image flashed:** 
- Not be able to flash any image.
- **What do you think should have happened:** 
- When I click the balenaEtcher-Portable-1.7.9.exe file, the program should start and I can use it.
- **What happened:**
- After I clicked the balenaEtcher-Portable-1.7.9.exe file, literally nothing happens, no window, no sound, no warning messages, nothing, like I never clicked.
- I tried using older version of balenaEtcher, run as administrator, restart the system, all didn't help.
- But my surface Pro 4 can run it without problem, my surface Pro 4 was run on older version of **Windows 10 Pro** (Version 1511), my theory is, maybe the program doesn't recognize **Windows 10 Pro for Workstation**? I've seen some other program sees the Windows 10 Pro for Workstation as unknown version of Windows 10.
- **Do you see any meaningful error information in the DevTools?** 
- Scene I never be able to open the program, surely I can't enter DevTools too.

lurch, 2022-04-26:
See also #1866

7217043955, 2022-05-02:
Some updates:
x64 operating system, i7-11700k and GeForce GTX TITAN X.
I checked task manager, no process of it shows up.
Compatible mode and -disable-gpu no help.
After I clicked it, my CPU fan will spin faster briefly.
Because I do really want to use balenaEtcher, I downloaded and installed the install version, interestingly, it runs with no problem, but I decided to check the DevTools anyway, despite I can use it without problem, it seems still have some complain. 1 error and 3 warnings:
Failed to load resource: net::ERR_FILE_NOT_FOUND
gui.js:46 [Axios v1.7.9] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future
console.<computed> @ gui.js:46
[Axios v1.7.9] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
console.<computed> @ gui.js:46
[Axios v1.7.9] Transitional option 'silentJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
console.<computed> @ gui.js:46
Then I tried portable again, still not running.

closed 2023-07-12 by dfunckt