balena-io/etcher #3750: Etcher 1.7.8 UI Disappears + Gets Stuck at 32% when Flashing

dreamsyntax, 2022-04-24:
- **Etcher version:** 1.7.8 (windows installed version x64)
- **Operating system and architecture:** Windows 10 21H2 (19044.1645) AMD64
- **Image flashed:** manjaro-kde-21.2.6-220416-linux515.iso
- **What do you think should have happened:** Flashing completes successfully, or an error message appears.
- **What happened:** After reaching 32%, BalenaEtcher's UI became blank (the application is NOT frozen/not responding, but the UI is gone). Progress does not continue.
[image]
### 
- **Do you see any meaningful error information in the DevTools?** No, showed as "disconnected, press Ctrl+R to reconnect", on pressing Ctrl+R balenaEtcher crashed to desktop and did not reopen.

The USB in question previously had Ubuntu LTS 22.04 flashed onto it with BalenaEtcher 1.7.8 (Linux x64 host). The device had 3 visible partitions in windows.
Reopening BalenaEtcher after the failed 32% (without disconnecting the target device) and re-flashing works successfully.

Will update if I can find a reliable way to reproduce.

savage0ne, 2022-10-15:
I’m having the same issue, except mine got stuck at 34% when the UI disappears. Just closed etcher out and then restarted the flashing process. Hopefully it’ll complete this time 🤞🏻
This is like the 5th hurdle I’ve encountered while trying to flash kali for raspberry pi onto an sd card with windows 11. I’m thinking some windows 11 patches need to take place. All were issues I was able to find reported on here, so it’s not as if they are new.

dreamsyntax, 2022-10-15:
Interesting... I was never able to reproduce but this is the exact issue

closed 2023-07-12 by dfunckt

dreamsyntax, 2025-03-25:
Closing as Etcher is not recommended anymore.