balena-io/etcher #3755: Unable to launch Etcher on Pop!_OS 22.04

ChiefGyk3D, 2022-05-01:
- **Etcher version:** 
v.1.7.9
- **Operating system and architecture:** 
Pop!_OS 22.04 Linux (Ubuntu fork)
- **Image flashed:** 
N/A
- **What do you think should have happened:** 
It should open
- **What happened:**
[Axios v1.7.9] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'silentJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
[223715:0501/095432.144955:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
/tmp/.mount_balenaZLUpJ3/balena-etcher-electron: line 10: 223715 Trace/breakpoint trap   (core dumped) "${script_dir}"/balena-etcher-electron.bin "$@"

- **Do you see any meaningful error information in the DevTools?** 
See above

mcraa, 2022-05-02:
looks similar, does this help:
https://github.com/balena-io/etcher/issues/3639#issuecomment-981052011 ?

ChiefGyk3D, 2022-05-17:
Yes it actually did help and now it works

closed 2022-05-18 by mcraa