balena-io/etcher #3760: Could not flash VHD virtual drive

Babyvox550, 2022-05-11:
- **Etcher version:** 1.7.9
- **Operating system and architecture:** X86 WIN11
- **Image flashed:** macOS Monterey 12.1 21C5021h Installer for OpenCore 0.7.5 and CLOVER 5141 and PE.DMG
- **What do you think should have happened:** I want to flash the DMG file to a VHD drive，But only physical drives are listed.
- **What happened:**
- 
[image]
[image]
[image]
[image]

How could I  flash dmg to VHD drives.

mcraa, 2022-05-11:
Why do you need to flash to a virtual drive? You should be able to mount your image directly as a drive.

korhojoa, 2022-05-15:
> Why do you need to flash to a virtual drive? You should be able to mount your image directly as a drive.

That's just it. The drive _is_ mounted, but it is not shown as a choice.
Mounted VHD:
[image]
Not listed in target selection:
[image]

lurch, 2022-05-16:
Presumably that's because https://github.com/balena-io-modules/drivelist only looks for physical drives and doesn't look for virtual drives? :man_shrugging:

liyuhanaljlj, 2024-02-27:
I also encountered the same problem, how to solve it? But I found an old version 1.5.5 that supports VHD! But it is no longer supported after 1.5.5. I hope the author can give a solution, thank you!

dangtu, 2024-08-12:
> I also encountered the same problem, how to solve it? But I found an old version 1.5.5 that supports VHD! But it is no longer supported after 1.5.5. I hope the author can give a solution, thank you!

Can I have your 1.5.5 version? I downloaded 1.5.5 from github repo but not detect VHDs.