balena-io/etcher #3761: Not opening in Ubuntu 22.04

DannySanx, 2022-05-16:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **What do you think should have happened:** 
- **What happened:**
- **Do you see any meaningful error information in the DevTools?**

survivor303, 2022-05-17:
Almost month from the latest build, no apt for debian based systems and now they not update electron to be compatible with anything. is this project dead?

meguroyama, 2022-05-30:
I can confirm this is an issue that's happening on 22.04 LTS as well.

andygikling, 2022-05-30:
In Ubuntu 22.04, as stated in a related thread, this does work for now: 

./balenaEtcher-1.7.9-x64.AppImage --no-sandbox --disable-gpu-sandbox --disable-seccomp-filter-sandbox

wagogamer, 2022-06-24:
After installing FUSE I was able to run the appimage perfectly.

**For example, on Ubuntu (>= 22.04):**

sudo add-apt-repository universe
sudo apt install libfuse2

**For example, on Ubuntu (<= 21.10):**

sudo apt install fuse libfuse2
sudo modprobe fuse
sudo groupadd fuse

user="$(whoami)"
sudo usermod -a -G fuse $user

**For example, on openSUSE:**

sudo zypper install fuse libfuse2

In order to use fusermount on OpenSUSE with the default (?) "secure" file permission settings (see /etc/permissions.secure), your login needs to be part of the trusted group. To add yourself, run

sudo usermod -a -G trusted `whoami`

Then log out and log in for the change to take effect.

**For example, on Fedora:**

dnf install fuse

Chevifier, 2022-09-09:
> In Ubuntu 22.04, as stated in a related thread, this does work for now:
> 
> ./balenaEtcher-1.7.9-x64.AppImage --no-sandbox --disable-gpu-sandbox --disable-seccomp-filter-sandbox
> 
> Using these switches is not recommended though. Frankly, I trust my AppImage and I just need the tool to work... so, no sandbox for me!

> In Ubuntu 22.04, as stated in a related thread, this does work for now:
> 
> ./balenaEtcher-1.7.9-x64.AppImage --no-sandbox --disable-gpu-sandbox --disable-seccomp-filter-sandbox
> 
> Using these switches is not recommended though. Frankly, I trust my AppImage and I just need the tool to work... so, no sandbox for me!

This worked prefectly for me. Ill add a txt file with it to remember

Lichundead, 2022-09-11:
LinuxMint 21 same error 

> Running the AppImage via terminal gives the following output:
> 
> [Axios v1.7.9] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future [Axios v1.7.9] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future [Axios v1.7.9] Transitional option 'silentJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null) libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null) libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null) [121535:0517/100918.052255:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye. /tmp/.mount_balenaIUPyWn/balena-etcher-electron: line 10: 121535 Trace/breakpoint trap (core dumped) "${script_dir}"/balena-etcher-electron.bin "$@"

It worked perfectly with this:

> In Ubuntu 22.04, as stated in a related thread, this does work for now:
>
> ./balenaEtcher-1.7.9-x64.AppImage --no-sandbox --disable-gpu-sandbox --disable-seccomp-filter-sandbox
>
> Using these switches [...]

kmasaryk, 2022-10-09:
FWIW, version 1.5.19-x64 still works without issue in Ubuntu 22.04.

AliAbukahil, 2022-10-27:
> After installing FUSE I was able to run the appimage perfectly.
>
> **For example, on Ubuntu (>= 22.04):**
>
> sudo add-apt-repository universe sudo apt install libfuse2
>
> **For example, on Ubuntu (<= [...]

I can confirm too, that this worked for me on 22.10. Thank you

david65536, 2024-02-20:
I can reproduce this issue on my ubuntu live usb, having run chmod +x on the .AppImage.  It seems fuse3 is installed natively on the ubuntu live. 

ubuntu@ubuntu:~/Downloads$ sudo apt-get install fuse
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'fuse3' instead of 'fuse'
fuse3 is already the newest version (3.10.5-1build1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
ubuntu@ubuntu:~/Downloads$ hostnamectl
 Static hostname: ubuntu
       Icon name: computer-desktop
         Chassis: desktop
      Machine ID: c6a6304c6a424bc5b4e0cc856fd2f8c8
         Boot ID: 865d131d927d4f419f8de8fa8b1ace69
Operating System: Ubuntu 22.04.3 LTS              
          Kernel: Linux 6.2.0-26-generic
    Architecture: x86-64

ubuntu@ubuntu:~/Downloads$ ./
balenaEtcher-1.18.11-x64.AppImage  squashfs-root/
ubuntu@ubuntu:~/Downloads$ ./balenaEtcher-1.18.11-x64.AppImage 
dlopen(): error loading libfuse.so.2

AppImages require FUSE to run. 
You might still be able to extract the contents of this AppImage 
if you run it with the --appimage-extract option. 
See https://github.com/AppImage/AppImageKit/wiki/FUSE 
for more information
ubuntu@ubuntu:~/Downloads$