balena-io/etcher #3763: Define .etch file format

phil-d-wilson, 2022-05-19:
A compressed file format with an open specification designed to wrap and augment standard image files with additional files, metadata and functionality in order to enhance the user experience. Designed for use with Etcher but available for use by all.

lurch, 2022-05-19:
This is kind of a duplicate of #707 ?
See also issues like #470 and #718 and #716 and #1113 and #1894

phil-d-wilson, 2022-05-19:
@lurch thank you for bringing all of that to my attention! I wasn't aware of this previous work. I'll chat to the team and consider which issue we'll progress this work with.

The good news is that we're working on this feature (again). :)

chrisys, 2022-06-02:
#852 also relevant perhaps