balena-io/etcher #3764: Not able to flash image located on NAS (smb)

nils-trubkin, 2022-05-19:
Hi,

Tried to flash a USB with an ISO that is located on a NAS (connected as attached network storage with windows smb). Got an error. Copying the image to the local drive solved it. Looks like a bug to me.

- **Etcher version: 1.7.9** 
- **Operating system and architecture: Win 11 x86_64** 
- **Image flashed: archlinux-2022.05.01-x86_64.iso** 
- **What do you think should have happened: Succesfull flash**
- **What happened: Got an error, had to copy the iso to local harddrive**
- **Do you see any meaningful error information in the DevTools? Yes, see below**

`Fri May 20 2022 01:05:38 GMT+0200 (Central European Summer Time) Write failed ({"image":{"size":866463744,"name":"archlinux-2022.05.01-x86_64.iso","hasMBR":true,"partitions":[{"offset":784334848,"size":81788928,"type":239,"index":1}],"extension":"iso","path":"N:\\Downloads\\OS\\archlinux-2022.05.01-x86_64.iso"},"drives":[{"size":32212254720,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive4","error":null,"isReadOnly":false,"displayName":"E:\\","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive4","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[{"path":"E:\\"}],"description":"SMI USB DISK USB Device","isUAS":false,"partitionTableType":null}],"driveCount":1,"uuid":"2571eb09-505c-4c89-80b9-edbf2b45b97f","status":"failed","flashInstanceUuid":"2571eb09-505c-4c89-80b9-edbf2b45b97f","error":{"name":"Error","code":"EUNPLUGGED","syscall":"open","errno":-4058},"sample":0.1,"applicationSessionUuid":"70a8a1d1-2025-4803-b753-15da64156e38","flashingWorkflowUuid":"17c4d3f2-c6cd-4098-8eb2-fb9b093b44b8"})`

lurch, 2022-05-22:
Known problem. See also e.g. #2858 (and the many other issues that it links to).

nils-trubkin, 2022-05-22:
So no fix no nothing for three years? Dead project or am I'm missing something?

nils-trubkin, 2022-05-22:
> Known problem. See also e.g. #2858 (and the many other issues that it links to).

So no fix no nothing for three years? Dead project or am I missing something?

XIVN1987, 2022-08-28:
I had the same problem.

nils-trubkin, 2022-08-28:
> I had the same problem.

Abandoned Electron project. Won't be fixed.

lurch, 2022-08-30:
https://github.com/balena-io/etcher/releases suggests that it isn't _that_ abandoned?

closed 2023-07-12 by dfunckt

nils-trubkin, 2023-07-12:
@dfunckt The issue is still present and relevant as of version 1.18.10
Why did you close an active bug report?

Please re-open.

dfunckt, 2023-07-12:
This is a duplicate of #2858.

nils-trubkin, 2023-07-13:
Thank you for your response