balena-io/etcher #3766: Unable to find pkexec or kdesudo.

davidak, 2022-05-28:
- **Etcher version:** 1.7.9
- **Operating system and architecture:** NixOS 22.05beta67.6efc186e607 (x86_64)
- **Image flashed:** https://hydra.nixos.org/build/178370308/download/1/nixos-gnome-22.05beta185.c5e45824dad-x86_64-linux.iso
- **What do you think should have happened:** No error
- **What happened:** When i click "Flash", i get an error.
- **Do you see any meaningful error information in the DevTools?** Can't enable

[image]

I have run it using a wrapper script since NixOS does not follow the Filesystem Hierarchy Standard (FHS).

```
[davidak@gaming:~]$ appimage-run Downloads/balenaEtcher-1.7.9-x64.AppImage --help
balenaEtcher-1.7.9-x64.AppImage installed in /home/davidak/.cache/appimage-run/bdb79c509cf7ea7ce3a33c767772fb142ec78c9f1b6266aac8d81e1c5c56024f
[Axios v1.7.9] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'silentJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
ready-to-show: 344.176ms
Checking for update
Update for version 1.7.9 is not available (latest version: 1.7.9, downgrade is disallowed).
```

It actually works when using sudo.

```
[davidak@gaming:~]$ sudo appimage-run Downloads/balenaEtcher-1.7.9-x64.AppImage
[sudo] password for davidak: 
balenaEtcher-1.7.9-x64.AppImage installed in /root/.cache/appimage-run/bdb79c509cf7ea7ce3a33c767772fb142ec78c9f1b6266aac8d81e1c5c56024f
[Axios v1.7.9] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'silentJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
ready-to-show: 447.447ms
Checking for update
Update for version 1.7.9 is not available (latest version: 1.7.9, downgrade is disallowed).
Error creating proxy: Unknown or unsupported transport “disabled” for address “disabled:” (g-io-error-quark, 13)
Error creating proxy: Unknown or unsupported transport “disabled” for address “disabled:” (g-io-error-quark, 13)
Error creating proxy: Unknown or unsupported transport “disabled” for address “disabled:” (g-io-error-quark, 13)
Error creating proxy: Unknown or unsupported transport “disabled” for address “disabled:” (g-io-error-quark, 13)
Error creating proxy: Unknown or unsupported transport “disabled” for address “disabled:” (g-io-error-quark, 13)

(balena-etcher-electron.bin:581248): dconf-WARNING **: 13:28:43.739: failed to commit changes to dconf: Unknown or unsupported transport “disabled” for address “disabled:”
```

mcraa, 2022-05-30:
Seems like we are not the first
https://github.com/NixOS/nixpkgs/issues/76526
The good news is there is a fix for that, thanks for the report.
We will include the fix in a future release.

raoel, 2023-06-24:
same error, Debian Testing
```$ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux trixie/sid
Release:        n/a
Codename:       trixie
```

rayluo, 2024-04-01:
For what it's worth, running the `balenaEtcher.AppImage` from command line seems to help. I ran into the same issue, when running the AppImage by double-clicking it from my desktop. Later I tried `sudo /path/to/balenaEtcher.AppImage`, it was even worse because the UI was largely empty. And then, I tried simply `/path/to/balenaEtcher.AppImage`, and it worked.

AleXoundOS, 2024-04-25:
> Later I tried `sudo /path/to/balenaEtcher.AppImage`, it was even worse because the UI was largely empty.

When I login as root (using `su`), then UI is visible.

fengyang0317, 2024-06-20:
`sudo apt install pkexec` solves the issue