balena-io/etcher #3767: Electron 12 used by etcher is EOL (end of life)

onny, 2022-05-29:
Electron 12 used by etcher is EOL (end of life). Installing it on NixOS results in a warning which is quite annoying. It would be great to update to a newer Electron version.

- **Etcher version:** latest
- **Operating system and architecture:** NixOS
- **Image flashed:** 
- **What do you think should have happened:** 
- **What happened:**
- **Do you see any meaningful error information in the DevTools?**

TheRealOne78, 2022-06-13:
It's also an issue with NVIDIA graphics drivers, electron12(and below) not working correctly with gpu sandboxing, leading to this error:`[21921:0613/173615.147591:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.`

The only way to work around is to use this as root or append `--no-sandbox` or `--disable-gpu-sandboxing` arguments to the etcher executable file, which is quite hard to understand for some rookie users.

Nvidia does not seem to be the actual problem as described here: https://github.com/NVIDIA/open-gpu-kernel-modules/issues/302

and electron devs are not planning to update electron12 since it's not suported anymore: https://github.com/electron/electron/issues/34522

rradar, 2022-12-01:
> Electron 12 used by etcher is EOL (end of life). Installing it on NixOS results in a warning which is quite annoying. 

It's not only "annoying" but a security nightmare. :bomb: 

Even critical CVE's are not fixed anymore because etcher builds their telemetry and advertising program on top of obsolete software - luckily the first package maintainers are getting rid of etcher already to protect their users. :+1:

PR #4132 (2023-12-21): Modernize build pipeline
Switch to electron-forge which is now officially adopted by the Electron project. Solves a bunch of issues along the way.

Issues related to experimental v1.18.13:

Closes #4138 
Closes #4129
Closes #4119 

Build related issues:

Closes #4136 
Closes #4125 
Closes #4103 
Closes #4087 
Closes #3767 
Closes #3728 
Closes #3653 
Closes #3400 
Closes #3292 
Closes #3269 
Closes #3263 
Closes #2870 
Closes #2826 

(Admittedly, some of the issues are linked opportunistically -- please reopen them if the issue is not actually solved)
closed 2023-12-21 by dfunckt