balena-io/etcher #3770: Only open with sudo permission

M4rQu1Nh0S, 2022-06-05:
- **Etcher version:** 1.7.9
- **Operating system and architecture:** amd64
- **Image flashed:** balena-etcher-electron_1.7.9_amd64.deb
- **What do you think should have happened:**  Balena don't asked for permitions after running it from shortcut created in launch menu.
- **What happened:** App not run without advice, system alert about crash bug.
- **Do you see any meaningful error information in the DevTools?** Not checked.

I've installed the balena etcher v1.7.9 amd64 deb format in Kubuntu 22.04, but the app not run via shortcut in launch menu not even ask for sudo password.

```
marcos@kubuntu-desktop:~/Downloads$ **/opt/balenaEtcher/balena-etcher-electron %U**

[Axios v1.7.9] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future

[Axios v1.7.9] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future

[Axios v1.7.9] Transitional option 'silentJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future

libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)

libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)

libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)

[10210:0605/130537.898818:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.

/opt/balenaEtcher/balena-etcher-electron: line 10: 10210 Trace/breakpoint trap   (core dumped) "${script_dir}"/balena-etcher-electron.bin "$@"
```

I has able to open balena etcher after open the shortcut file and search for the launch command, so i used the comando below with sudo and the balena etcher is now working.

```
marcos@kubuntu-desktop:~/Downloads$ **sudo /opt/balenaEtcher/balena-etcher-electron %U**
[Axios v1.7.9] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'silentJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
ready-to-show: 612.487ms
Error creating proxy: Unknown or unsupported transport “disabled” for address “disabled:” (g-io-error-quark, 13)
Error creating proxy: Unknown or unsupported transport “disabled” for address “disabled:” (g-io-error-quark, 13)
Error creating proxy: Unknown or unsupported transport “disabled” for address “disabled:” (g-io-error-quark, 13)
Error creating proxy: Unknown or unsupported transport “disabled” for address “disabled:” (g-io-error-quark, 13)
Error creating proxy: Unknown or unsupported transport “disabled” for address “disabled:” (g-io-error-quark, 13)

(balena-etcher-electron.bin:10348): dconf-WARNING **: 13:05:48.522: failed to commit changes to dconf: Unknown or unsupported transport “disabled” for address “disabled:”

(balena-etcher-electron.bin:10348): dconf-WARNING **: 13:05:48.545: failed to commit changes to dconf: Unknown or unsupported transport “disabled” for address “disabled:”

(balena-etcher-electron.bin:10348): dconf-WARNING **: 13:05:51.009: failed to commit changes to dconf: Unknown or unsupported transport “disabled” for address “disabled:”

(balena-etcher-electron.bin:10348): dconf-WARNING **: 13:05:53.030: failed to commit changes to dconf: Unknown or unsupported transport “disabled” for address “disabled:”

(balena-etcher-electron.bin:10348): dconf-WARNING **: 13:05:53.570: failed to commit changes to dconf: Unknown or unsupported transport “disabled” for address “disabled:”

(balena-etcher-electron.bin:10348): dconf-WARNING **: 13:05:53.571: failed to commit changes to dconf: Unknown or unsupported transport “disabled” for address “disabled:”
(node:10348) Warning: No such label 'ready-to-show' for console.timeEnd()
(Use `balena-etcher-electron.bin --trace-warnings ...` to show where the warning was created)
```

vipulgupta2048, 2022-06-08:
Suggestion: if you intend to run etcher with sudo permission interactively, then how about appending the command with `askpass` that will prompt you for the password to run etcher with sudo permission. I use etcher's Appimages using Linux and I get an interactive window to that asks for password. Check out this [post](https://mixstersite.wordpress.com/2020/10/10/appimages-xubuntu-whisker-menu-ubuntu-applications/) and if it helps you out.

lurch, 2022-06-08:
It looks like your problem (when _not_ running with `sudo`) is actually "GPU process isn't usable." - and there's other issues discussing that https://github.com/balena-io/etcher/search?q=%22GPU+process%22&type=issues

ohhai, 2022-10-13:
Root access should not be asked when not necessary.
I did `chown 0:user /dev/sdb*` so disk device is writeable by user, but Etcher still asks me for root auth.
Partially related: #3766, #3791