balena-io/etcher #3771: v1.7.9 regression on debian installation directory

kb2ma, 2022-06-08:
- **Etcher version:** 1.7.9
- **Operating system and architecture:** Ubuntu 20.04 
- **Image flashed:** N/A
- **What do you think should have happened:** 

1. Run installation script for Debian package infrastructure from cloudsmith.io.
2. Install balena-etcher-electron package
3. `postinst` script references correct installation directory `/opt/balenaEtcher` and creates symlink `/usr/bin/balena-etcher-electron`

- **What happened:**

Error message during installation:

```
Setting up balena-etcher-electron (1.7.9) ...
chmod: cannot access '/opt/balena-etcher-electron/chrome-sandbox': No such file or directory
```
Symlink `/usr/bin/balena-etcher-electron` is broken

A look inside the `balena-etcher-electron_1.7.9_amd64.deb` file shows the postinst script is incorrect.

- **Analysis**

I built the deb package from the master branch here, and it builds and installs correctly. So something broke in the process of getting the right package to/from cloudsmith.io.

mcraa, 2022-06-10:
I already suspected the CI and started working on gh actions, which will build the same way you do locally.

beej, 2022-06-14:
Debian n00b here.

What's the workaround?

I'd be thrilled if @kb2ma or somebody else knowledgeable were to provide instructions for fixing the postinst or for coaxing Debian into installing a working older .deb.

kb2ma, 2022-06-14:
@beej to workaround you can manually run the steps that should be in the `postinst` file. See the [definition](https://github.com/balena-io/etcher/pull/3735/files) of `after-install.tpl` in #3735.

jorgy80236, 2022-07-12:
I am having the same issue, but on Rocky 8.6 (RedHat / CentOS clone).  I used the balena-etcher_rpm.sh script to install the repo, and then used dnf to install balena-etcher-electron-1.7.7-1.x86_64 from the repo.   The result is the same as kb2ma has on Ubuntu.

/usr/bin/balena-etcher-electron: broken symbolic link to /opt/balena-etcher-electron/balena-etcher-electron

where it should be a link to /opt/balenaEtcher/balena-etcher-electron

Thanks.