balena-io/etcher #3777: I can´t open balena etcher in Ubuntu 22.04

daxtas, 2022-06-18:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **What do you think should have happened:** 
- **What happened:**
- **Do you see any meaningful error information in the DevTools?**

daxtas, 2022-06-18:
I downloaded Appimage. It does not open. It is not a permissions issue.
I installed APT, it does not open.
What I can do?

pdiasgon, 2022-06-20:
Hello,
Thank you for your great product.

Unfortunately, I've the same problem.
Here is the result when i launch balena etcher from the terminal:

paul@p-desk:~/Applications$ ./balenaEtcher-1.7.9-x64.AppImage 
[Axios v1.7.9] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'silentJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
[3534187:0620/162602.217692:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
/tmp/.mount_balenaFRNGw5/balena-etcher-electron : ligne 10 : 3534187 Trappe pour point d'arrêt et de trace              (core dumped) "${script_dir}"/balena-etcher-electron.bin "$@"

Regards
Paul

pdiasgon, 2022-06-20:
Thank's

Erol444, 2022-06-20:
I tried running `./balenaEtcher-1.7.9-ia32.AppImage ` on Ubuntu 20.04. Error below. I already have fuse and libfuse2 installed.
```
dlopen(): error loading libfuse.so.2

AppImages require FUSE to run. 
You might still be able to extract the contents of this AppImage 
if you run it with the --appimage-extract option. 
See https://github.com/AppImage/AppImageKit/wiki/FUSE 
for more information```

augustbassfa, 2022-06-21:
Buenos días comunidad! Saludos me pasa este error 
```
augustbassfa@augustbassfa-desktop:~$ sudo apt install balena-etcher-electron
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias... Hecho
Leyendo la información de estado... Hecho
E: No se ha podido localizar el paquete balena-etcher-electron
```
Ya intente instalar con la AppImage y tampoco funciona. ¿que debo hacer?

de ante mano gracias

wmthornton, 2022-07-01:
> I tried running `./balenaEtcher-1.7.9-ia32.AppImage ` on Ubuntu 20.04. Error below. I already have fuse and libfuse2 installed.
>
> ```
> dlopen(): error loading libfuse.so.2
>
> AppImages require FUSE to [...]

If you install FUSE, make sure you install FUSE3 and not the standard FUSE. There is an issue with FUSE where it will remove your Gnome icons from the desktop and the option to manage them due to an incompatibility with Ubuntu 22.04 and the updated GNOME libraries. Just my $.02

lidarbtc, 2022-07-05:
use ```./balenaEtcher-1.7.9-x64.AppImage --no-sandbox```

It's work for me.

ollinurb, 2022-07-23:
Experiencing same issue as @Erol444.

`./balenaEtcher-1.7.9-x64.AppImage --no-sandbox
dlopen(): error loading libfuse.so.2

AppImages require FUSE to run. 
You might still be able to extract the contents of this AppImage 
if you run it with the --appimage-extract option. 
See https://github.com/AppImage/AppImageKit/wiki/FUSE 
for more information
`
Any way to solve the issue without installing FUSE as it is not recommended for Ubuntu 22.4?

lurch, 2022-07-23:
> Any way to solve the issue without installing FUSE as it is not recommended for Ubuntu 22.4?

You could use the deb package instead of the AppImage?

ollinurb, 2022-07-23:
> > Any way to solve the issue without installing FUSE as it is not recommended for Ubuntu 22.4?
> 
> You could use the deb package instead of the AppImage?

Following the steps provided at https://github.com/balena-io/etcher#packages (adding Etcher Debian repository and installing) I get an access Icon on the Applications window. But it doesn't open the program when clicked.

wmthornton, 2022-07-25:
You can simply use 'apt' to install FUSE3. I had this issue on 22.04 and it fixed it.

ollinurb, 2022-07-25:
FUSE3 comes installed onto 22.04. Running 'apt' to install it just confirms that the package is already there.

[Checking the issues of AppImage](https://github.com/AppImage/AppImageKit/issues/1120) I decided to install libfuse2.

```
sudo apt install libfuse2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  gconf2 libgdk-pixbuf-xlib-2.0-0 libgdk-pixbuf2.0-0
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  libfuse2
0 upgraded, 1 newly installed, 0 to remove and 17 not upgraded.
Need to get 90,3 kB of archives.
After this operation, 330 kB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu jammy/universe amd64 libfuse2 amd64 2.9.9-5ubuntu3 [90,3 kB]
Fetched 90,3 kB in 2s (43,9 kB/s)                      
Selecting previously unselected package libfuse2:amd64.
(Reading database ... 248325 files and directories currently installed.)
Preparing to unpack .../libfuse2_2.9.9-5ubuntu3_amd64.deb ...
Unpacking libfuse2:amd64 (2.9.9-5ubuntu3) ...
Setting up libfuse2:amd64 (2.9.9-5ubuntu3) ...
Processing triggers for libc-bin (2.35-0ubuntu3) ...

```

Now after running the AppImage I get this error message:
```
./balenaEtcher-1.7.9-x64.AppImage
[Axios v1.7.9] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'silentJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[130370:0725/120015.005621:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
/tmp/.mount_balenauOrebc/balena-etcher-electron: line 10: 130370 Trace/breakpoint trap   (core dumped) "${script_dir}"/balena-etcher-electron.bin "$@"
```

mcraa, 2022-07-25:
for the last one check #3639

ollinurb, 2022-07-25:
> for the last one check #3639

Thanks. Running `./balenaEtcher-1.7.9-x64.AppImage --disable-gpu-sandbox ` I can successfully open balenaEtcher.

pablofl, 2022-08-01:
as workaround:
- sudo nautilus
- click the Appimage

pablofl, 2022-08-01:
as workaround:
- sudo nautilus
- run the app works here

KamilGaribov, 2022-11-26:
Enable "Execute" from properties/permissions of balenaEtcher.AppImage

peliopoulos, 2022-12-21:
Possible workaround for gnome at least:
https://github.com/balena-io/etcher/issues/3809#issuecomment-1350200863

EnzoCamilo, 2023-01-07:
> use `./balenaEtcher-1.7.9-x64.AppImage --no-sandbox`
> 
> It works for me.

it's not working for me

ManuKriz, 2023-02-04:
https://github.com/balena-io/etcher/releases/download/v1.5.122/balenaEtcher-1.5.122-x64.AppImage

ManuKriz, 2023-02-16:
https://github.com/balena-io/etcher/releases/download/v1.5.122/balenaEtcher-1.5.122-x64.AppImage

Using this version fixed my issue

[source](https://askubuntu.com/questions/1363783/cant-run-an-appimage-on-ubuntu-20-04)

But if you are trying to create bootable drive of Windows on Ubuntu use WoeUSB
https://github.com/WoeUSB/WoeUSB/releases

dfunckt, 2023-07-12:
Likely duplicate of #3761? Closing.

closed 2023-07-12 by dfunckt