balena-io/etcher #3778: Failure to flash from balena dashboard

AndreiGr, 2022-06-22:
- **Etcher version:** 1.5.115
- **Operating system and architecture:** Windows 10 (x64)
- **Image flashed:** Raspberry Pi 4
- **What do you think should have happened:** After I click on the flash button in balena dashboard I expected balena etcher to open and start flashing the correct image on the disk.
- **What happened:** After I click on the flash button in balena dashboard, balena etcher opens and shows the following error:
[image]

- **Do you see any meaningful error information in the DevTools?** 
`{"stack":"Error: ENOENT: no such file or directory, open 'C:\\WINDOWS\\system32\\%7B%22method%22%3A%22POST%22%2C%22url%22%3A%22https%3A%2F%2Fapi.balena-cloud.com%2Fdownload%3FappId%3D1944541%26deviceType%3Draspberrypi4-64%26version%3D2.98.33[...]`

closed 2022-06-23 by mcraa

AndreiGr, 2022-06-23:
Ok, good to know.
Do you have any idea when a newer pre-made installer will be made for Windows?

Because from what I see, the latest I can download is 1.5.115

mcraa, 2022-06-23:
Where do you see 1.5 as latest?

https://github.com/balena-io/etcher/releases/download/v1.7.9/balenaEtcher-Setup-1.7.9.exe 
here you have the real latest from the [Releases](https://github.com/balena-io/etcher/releases/tag/v1.7.9) in github.
etcher.io (or balena.io/etcher) download button points to the same file.

AndreiGr, 2022-06-23:
On the [main ](https://www.balena.io/etcher/) page:
[image]

mcraa, 2022-06-23:
Well, that is weird, thanks for pointing it out.

AndreiGr, 2022-06-23:
No worries. Thanks for helping me with the [Releases](https://github.com/balena-io/etcher/releases/) link. I tried to find it myself, but what I found redirected me to the main page, haha.
Have a great day!

lurch, 2022-06-23:
The Etcher website sometimes points to an old version if you have JavaScript turned off? (which has happened in several issues here previously)

mcraa, 2022-06-23:
Cross-site policy blocked to fetch the right version, and it fallbacks to 1.5 
I guess browsers started to handle this more strictly @lurch cause the policy was the same for a long time.
The site was redeployed it should show the right version now.