balena-io/etcher #3784: Telemtry should be opt-in

oliversalzburg, 2022-07-13:
- **Etcher version:** 1.7.9
- **Operating system and architecture:** Windows x64
- **Image flashed:** irrelevant
- **What do you think should have happened:**  No information is sent to third-parties without my consent
- **What happened:** You require me to discover and opt-out of telemetry myself and happily send everything you want prior to that.
- **Do you see any meaningful error information in the DevTools?** No

Given that your product is recommended by Canonical as the way to put their images on storage media, I was unpleasantly surprised to find opt-out telemetry in your product. If you want to exfiltrate information from my machine, please ask me before doing so, so that I can then use a different product to flash images onto media.

rradar, 2022-08-14:
@oliversalzburg wrote:
> Given that your product is recommended by Canonical

Said to hear that it still recommended - many others (like the raspberry trading ltd) luckily stopped already promoting this software.

Your issue btw. will not be addressed as balena knows already long time that they somewhat not compatible with eu rights.

Just a few of the issues mentioning the gdpr in the [issue tracker](https://github.com/balena-io/etcher/issues?q=gdpr): https://github.com/balena-io/etcher/issues/3476, https://github.com/balena-io/etcher/issues/3006, https://github.com/balena-io/etcher/issues/2977, https://github.com/balena-io/etcher/issues/2914, https://github.com/balena-io/etcher/issues/2890, https://github.com/balena-io/etcher/issues/2766, https://github.com/balena-io/etcher/issues/2599, https://github.com/balena-io/etcher/issues/2497

> that I can then use a different product to flash images onto media.

If you are okay to not use a 0.5gb monster with ad/spy/etc-ware included but a simple yet powerful tiny (lesss than 0.5mb) program which just does it's job I can suggest [USBImager :arrow_down:](https://gitlab.com/bztsrc/usbimager)

oliversalzburg, 2022-08-14:
Very naughty... And, I guess, I should have searched for other issues mentioning this subject.

Thanks for your advice! I just went with Rufus and then I went with Debian instead of Ubuntu. I really don't see the point in downloading a huge Electron app to write images. Stuff like etcher has no place in the Linux world that I know. Especially given the absolute lack of responsibility as demonstrated on this issue tracker.

sunknudsen, 2023-01-18:
As a privacy and security researcher and YouTuber, I have [published](https://sunknudsen.com/privacy-guides?search=sudo%20dd%20if%3D) many guides that involve flashing operating systems to flash drives and SD cards.

I totally agree with the following statement… I am always worried when publishing guides that use `dd` as one can easily destroy data.

> Here at [balena](https://balena.io/) we have thousands of users working through our getting started process and until recently we were embarassed about the steps that involved flashing an SD card.

Thankful for Etcher which makes process less dangerous and more accessible to non-technical users. 🙌

That said, I use [Little Snitch](https://www.obdev.at/products/littlesnitch/index.html) to monitor apps calling home and was mind blown when I realized how many times Etcher calls home (even once update checks and telemetry had been disabled).

Please make telemetry and any other calling home (including sentry.io monitoring) opt-in.

sunknudsen, 2023-01-18:
[Raspberry Pi Imager](https://www.raspberrypi.com/software/) is a more privacy-conscious alternative (see [telemetry](https://github.com/raspberrypi/rpi-imager#telemetry)).

That said, Etcher nailed the UX… perhaps Etcher could implement a flag to disable telemetry like Raspberry Pi Imager did.

Even better… telemetry could be opt-in.

mcraa, 2023-01-18:
> many others (like the raspberry trading ltd) luckily stopped already promoting this software.

luckily they still find balena useful as they still use [balena packages](https://github.com/raspberrypi/rpi-imager/blob/qml/src/dependencies/drivelist/src/drivelist.cpp).

Thanks for the thoughtful words @sunknudsen , as you see with releases 1.14+ the logging part is under complete revamp.
All data were anonymized before, and that will be more visible from now on. UI changes to opt-in could come down the road.

On your second comment about RPi Imager, I am a bit surprised, they have also opt-out, more hidden than Etcher. You need to select an image first, so the config button in the lower left appears where you need to scroll to the bottom to remove the tick.
The data they collect written on the link you added is really understandable, etcher did the same. On top of that, behavior on the UI and errors. Changing the configuration through the command line or other editor, etcher too can be started the first time already with analytics disabled. 
You are right our documentation is not that good, not like many people read it. 😅  If they would read we would have much fewer issues repeating the same. And if you read those issues "it does not work!" without any useful details, I hope you see how helpful or life-saving some error logs are.

I liked your guides and videos, they are really good quality, organized and understandable 👍

sunknudsen, 2023-01-18:
Thanks for joining conversation @mcraa.

> Changing the configuration through the command line or other editor, etcher too can be started the first time already with analytics disabled.

Promising… how? That should be surfaced on [README.md](https://github.com/balena-io/etcher/blob/master/README.md).

> And if you read those issues "it does not work!" without any useful details, I hope you see how helpful or life-saving some error logs are.

Feel you… I also use Sentry. That said, great apps should always ask users before submitting bug reports that typically include sensitive data such as the user’s IP, OS, and, often more… Sentry is by default very data hungry (which, arguably, can be circumvented by privacy-conscious developers).

mcraa, 2023-01-18:
> 
> Promising… how? That should be surfaced on [README.md](https://github.com/balena-io/etcher/blob/master/README.md).
> 

When you toggle it on the UI, it is saved into a `config.json` file in the [userData](https://www.electronjs.org/docs/latest/api/app#appgetapppath) directory (different path for different OSes).
Editing that file setting `errorReporting: false`, or removing analytics related properties from the same or package.json as accessed in the [corresponding module](https://github.com/balena-io/etcher/blob/master/lib/gui/app/modules/analytics.ts#L139)

Scr3amer, 2023-09-09:
Guys can you chill for a second ?

I was reading:
https://github.com/balena-io/etcher/issues/2977

I quote [@sneak](https://github.com/sneak) "The data Etcher transmits from my machine includes my IP address, which uniquely identifies me"

You guys do not understand internet for first. IP address does not always identify you uniquely, for especially IPv4 because you are in a package with other customers from that ISP.

**Also just going on github posting a post is giving your IP to Microsoft.**

I understand that for a free software with no license management telemetry is just for statistics and you don't want to be in any stats but stop calling it spying if the only payload is IP address. I also understand the principle of asking authorization for everything but don't bend the truth by saying it is spying when it is not because you become also dishonest for people claiming to be ethics fighters.

Hell even if the payload contains hardware models it is not spying because knowing you have an intel or amd cpu won't change a thing regarding your ID. Spying is uploading user data like your pictures or passwords and such. Anything else is not spying, it is data collection but not spying, learn the difference.

As long as Etcher explicitely says what they collect and there is no user generated info it is not spying.

Also just use a firewall and block the app. Or modify source code or use an other software if you are not happy.

In other words: don't exagerate claims.

Scr3amer, 2023-09-13:
I agree that your post was less emotional than the one I linked but if IP is the only payload gathered by Balena then there is no exfiltration since it is generated by your ISP not by you.

I didn't read the code of Balena but since you created the post you probably did the work: what does Balena exfiltrate exactly?

Because if it is an empty POST ping there is nothing besides whats in the frame of a HTTP packet.

honobo, 2023-09-17:
This is not a question of “opt in” or “opt out”: we have to do the morally right thing. While this is probably not going to be a popular opinion here, I will say this respectfully in good conscience [who I am should be irrelevant to the following statements].

[Reject](https://www.gnu.org/philosophy/words-to-avoid.html#OptOut) abusive or otherwise ethically questionable software, free or nonfree, and warn and help out your friends, your community, about that too. Which is trivially easy when there are better free options already. It’s also important for one to make donations as much as possible to support whatever work they like, so that its author may not need to make them adware or spyware.

@Scr3amer 

> Also just going on github posting a post is giving your IP to Microsoft. So stop crying for IP. IP is public information not private.

That is not quite true. Currently one can sign up and use Github exclusively on Tor—browsing with Tor Browser, and otherwise via Socks5 proxy (when signing up, use an anonymous email address which itself can be signed up and used exclusively on Tor). How long Github will allow privacy-oriented users is anyone’s guess. Soon or later they’re likely to become a garden with higher walls with the pretext of security or something, possibly even requiring your biometric info. *Freedom to study and modify* codes are two of our core values, and in principle free activities, by definition, should be able to be engaged freely by anyone, without showing a government-issued photo ID or anything like that.

I trust that anyone will agree that Github provided by Micro$oft is not exactly an ideal place for free software—at least spiritually. This situation has been questioned since long time ago and programmers with good intuition, or who need privacy, are indeed not using Github. An exception could be an intentional free-speech test as in Tornado Cash (once kicked out and now fighting back supported by eff), but that is a “meta-use” for demonstration i.e. not like they really want to use Github in an oridnary sense.

Escape from “them” before it’s too late.

aethernet, 2023-10-13:
We're taking your suggestion into consideration.
Note that the telemetry is anonymize on both the client and server and we don't store PII.

closed 2023-10-13 by aethernet

embedding-shapes, 2025-10-30:
> Tails now recommends not to use Etcher as a result of this:

New working link: https://tails.net/news/rufus/

> Shortly after, balenaEtcher started displaying ads. Although we didn't like that, we initially didn't view it as a significant privacy risk and had no better alternative at the time.

> However, in 2024, the situation changed: balenaEtcher started sharing the file name of the image and the model of the USB stick with the Balena company and possibly with third parties.