balena-io/etcher #3785: BalenaEtcher flatpak version

farribeiro, 2022-07-14:
I wrote a balena-etcher flatpak yaml based on deb package, is working. take a look the file in https://gist.github.com/farribeiro/b26f9dacbf3e04bbe94fbede07b0808b

Note: need write appdata if you want publish.

To appreciation,
Best Regards,
Fábio

lurch, 2022-07-15:
Related issues: https://github.com/balena-io/etcher/search?q=flatpak&type=issues

lurch, 2022-07-16:
I'm not directly involved with this project, I was merely pointing out that I'd seen other flatpak-related issues pop up in this project before :slightly_smiling_face:

farribeiro, 2022-07-16:
I can reopen as reminder the problems

diogopessoa, 2026-09-24:
Hi, 
The way Etcher currently asks for permission to write/unmount disks would need to be adapted to work with Flatpak's sandboxing model. 

Two possible approaches to make Etcher's disk write/unmount work under Flatpak:

1. Route the existing privilege-escalation call through `flatpak-spawn --host pkexec <helper>` instead of invoking it from inside the sandbox.
2. Rework the mount/write logic to use udisks2 over D-Bus, which works natively inside the sandbox without needing host access.

Option 1 is likely the smaller change given the current code structure; option 2 is more sandbox-native but requires more rework.

Thanks and peace ✌🏻
(I am not a developer)