balena-io/etcher #3792: New Pi Compute Modules Not Initializing

bhjelstrom, 2022-08-02:
- 1.7.9
- Ubuntu 20.04, Win10
- No Dev Tools output

I have tested 7 new CM4s with HW revision ```c03141``` and new bootloaders. They function as expected only when I run the latest rpiboot manually. Assuming https://github.com/balena-io-modules/node-raspberrypi-usbboot needs an update.

lsusb:
```Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0a5c Broadcom Corp.
  idProduct          0x2711 
  bcdDevice            0.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
```

mcraa, 2022-08-02:
Thanks for sharing, we are working on it already.

aessig, 2022-08-03:
@mcraa What should we do in the meantime?

mcraa, 2022-08-04:
Sorry for the long reply, I don't know what you should do I only know what workaround is possible, similar to what we are working on.
The CM4s should be a 2711 device. So you can clone the repos and hardcode it in 
[node-raspberrypi-usbboot](https://github.com/balena-io-modules/node-raspberrypi-usbboot/blob/85941a6b64f7ac86070fdb53e0ff866950737a46/lib/index.ts#L525) to always detect 2711.
Then update it (locally) in the etcher-sdk, and update the etcher-sdk in etcher and build etcher for yourself.
This version of etcher then would work only work with cm4 and cm4s.
Or you can run your updated sdk directly like [ts-node examples/usb-boot.ts <..args>](https://github.com/balena-io-modules/etcher-sdk/blob/master/examples/usbboot.ts) without the struggle of building etcher.

It will probably take us this or the next week at least to test everything and make sure we don't break the support of any already supported devices.

aessig, 2022-08-05:
@mcraa Thanks for your reply. Will try it.

tosuns, 2022-08-24:
What is the current status of this issue? Do you guys have an idea when you will release a new version which contains a fix this issue ?

fix3000, 2022-08-24:
Also here to poke. I can use rpiboot to work around the issue for now, but would be good to have the direct route back. It might be worth mentioning that "older" CM4s with a 2020 manufacturing date are directly recognized by balenaEtcher 1.7.9. It's the newer ones with a 2021 mfg date that I'm seeing the issue with. Suspecting there may be a newer bootloader when they roll off the line that's causing the issue. 

This may not be the right place for this, but would any update also be usable on Etcher Pro? 

I can do some limited testing if that's helpful; feel free to reach out.

bhjelstrom, 2022-08-25:
I can confirm what @fix3000 is saying. To be clear, we are seeing this with 2021 and newer "CM4" units. There is a new "CM4-s" model that may or may not be detected, but I haven't tested the "CM4-s".

Shipping CM4 based products now so bumping again.

mcraa, 2022-08-25:
Well, that name is confusing. CM4s meaning plural CM4, and CM4S as the new small format devices.
Seeing "New Pi" and "CM4s" I thought about the new small devices, it is good to point out there are "normal" size devices which have the issue.
cc @zwhitchcox

lurch, 2022-08-25:
We've not added it to our [general documentation](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#new-style-revision-codes) (because CM4S is a [limited-distribution product](https://github.com/raspberrypi/documentation/pull/2608#issuecomment-1219469656)), but the CM4S has a `Type` code of 0x15 https://github.com/raspberrypi/documentation/pull/2557 (regular CM4 has `Type` code 0x14).

https://forums.raspberrypi.com/viewtopic.php?t=339279 has more info about the CM4 with revision code `c03141`.

tosuns, 2022-08-25:
@lurch What is the meaning of "limited-distribution product" and what kind of implication does it have with lifetime and support ?

lurch, 2022-08-25:
As it says on the [CM4S product page](https://www.raspberrypi.com/products/compute-module-4s/):
> Based on the Raspberry Pi 4 Model B architecture, Compute Module 4S is intended for specific industrial customers migrating from Compute Module 3 or Compute Module 3+ and is not for general sale. For new customers who are designing products, we recommend using Raspberry Pi Compute Module 4.

fix3000, 2022-09-27:
I haven't seen any recent update here on this issue, but is it something that's being worked on? Or planned to be worked on?

We're reaching a decision point on how we're going to flash all these new CM4 devices we'll soon be getting. If we need to work out an alternate plan, that's ok, just want to get ahead of it.

zwhitchcox, 2022-09-29:
Hey, we think we have identified the problem, but still needs more testing. I will post an update here hopefully very soon.

fix3000, 2022-09-30:
That's good to hear, hope it progresses well. Thanks.

iamk3, 2022-10-07:
> Hey, we think we have identified the problem, but still needs more testing. I will post an update here hopefully very soon.

If you have a beta, I'd love to test it out. I'm bummed at the moment not being able to flash my two 32GB CM4.

fix3000, 2022-10-07:
> > Hey, we think we have identified the problem, but still needs more testing. I will post an update here hopefully very soon.
> 
> If you have a beta, I'd love to test it out. I'm bummed at the moment not being able to flash my two 32GB CM4.

FYI, you can use rpiboot first to get the module mounted as storage and then use balenaEtcher to flash it.

I'd also be open to testing a beta if one is available.

bhjelstrom, 2022-10-08:
Also applies to a03141 units

iamk3, 2022-10-10:
> FYI, you can use rpiboot first to get the module mounted as storage and then use balenaEtcher to flash it.
>
> I'd also be open to testing a beta if one is available.

Would you mind listing the steps to do this? I can't seem to figure it out. I have tried both running rpiboot and then plugging in the cm4 and plugging in the cm4 then running rpiboot.

lurch, 2022-10-10:
AFAIK it _should_ just be a case of following the instructions [here](https://www.raspberrypi.com/documentation/computers/compute-module.html#setting-up-the-cmio-board), but then using balenaEtcher instead of `dd` or Raspberry Pi Imager? :man_shrugging:

zwhitchcox, 2022-10-20:
Hey, thanks for the community support and willingness to test. We actually have an internal tester with the required hardware, so that won't be necessary. We have a working version of the low level model. Working on getting an Etcher release right now. Just wanted to give an update.

Thank you for your patience!!

fix3000, 2022-10-20:
That's fantastic news, thanks so much for looking at this.

Relatedly, when the release is, well... released, would it also be
available for those of us that have Etcher Pro?

Thanks again.

zwhitchcox, 2022-10-20:
Great question @fix3000 We are currently working on automating the process, but for now, I can go ahead and just manually change push the changes to EtcherPro repo.

zwhitchcox, 2022-10-20:
Oops, didn't mean to close this yet.

iamk3, 2022-11-07:
Just out of curiosity, is there any update on expected release timeframe?

bhjelstrom, 2022-11-08:
https://github.com/balena-io/etcher/actions/runs/3423085558
Watching this PR in real time but has failed a few times now :)

vrajesh-patel03, 2022-11-11:
I am waiting for this fix too.

zwhitchcox, 2022-11-12:
You can try this out now in version [`1.10.0`](https://github.com/balena-io/etcher/releases/tag/v1.10.0) I do not have a CM4, so I haven't been able to test it myself though. Would someone mind letting me know if it is working?

bhjelstrom, 2022-11-12:
@zwhitchcox Not working, but I'm pretty sure etcher-sdk needs to be updated to 7.4.2. The PR which upgrades etcher-sdk to 7.4.2 is still failing

https://github.com/balena-io/etcher/pull/3848

https://github.com/balena-io-modules/etcher-sdk/commits/master

vrajesh-patel03, 2022-11-12:
I can also confirm that it is still not working.

iamk3, 2022-11-17:
I downloaded and installed 1.10.0. I was able to flash a CM4 with 32GB emmc once I opened rpiboot. I opened etcher, loaded the image to flash, started rpiboot, plugged in the CM4, and it initialized and flashed the image. It never successfully verified, but the flash was successful.

vrajesh-patel03, 2022-12-02:
@iamk3 Using rpiboot works fine if you're imaging one module at a time. It's when you initialize/mount as USB drive using Etcher that doesn't work with new CM4 modules. I have been using Etcher to image multiple CM4 modules. With the new CM4 modules, I have to run rpitool and can only image one module at a time. I don't know any other imager that can mount multiple CM4 modules and image multiple targets at same time.

lurch, 2022-12-29:
I don't work on Etcher myself, but I just noticed that https://github.com/balena-io/etcher/releases/tag/v1.13.0 claims to (finally?) fix the CM4 issues in Etcher.

mcraa, 2022-12-29:
> I don't work on Etcher myself, but I just noticed that https://github.com/balena-io/etcher/releases/tag/v1.13.0 claims to (finally?) fix the CM4 issues in Etcher.

~~claims~~ intends to
but yes that build has the latest etcher-sdk

mcraa, 2023-01-11:
https://github.com/balena-io/etcher/releases/tag/v1.13.3
Tested with cm4 on windows and mac.
(just to be quicker once than @lurch 😅 )

timg236, 2023-01-14:
The embedded start4.elf is out of date and won't work on the updated CM4. Patching the files post-install resolves this.

The update was released to Github Jul2021 so I'd recommend modifying the Etcher build-flow to pull in the usbboot as a Git submodule
https://github.com/raspberrypi/usbboot/tree/master/msd

Raspberry Pi provide a tag for each APT release so Etcher can track that.

vrajesh-patel03, 2023-01-16:
Tested by swapping start4.elf and I can see new CM4 modules mounted as a USB drive. 👍  Would be nice to have it incorporated in the next release.

lurch, 2023-01-16:
I'm **not** going to steal @mcraa 's thunder this time... :laughing:

mcraa, 2023-01-17:
No problem @lurch my 3 seconds of fame was enough.
https://github.com/balena-io/etcher/pull/3980 is already there

mcraa, 2023-01-17:
merged => https://github.com/balena-io/etcher/releases/tag/v1.14.2

mcraa, 2023-01-17:
It was added to EtcherPro https://github.com/balena-io-hardware/etcherPro-fleet-sw/pull/32#pullrequestreview-1251958682
Release `ccda501` version 1.3.0 is already finalized, and 16 EtcherPros already updated to it.

mcraa, 2023-01-23:
Thanks for the heads up @bhjelstrom must have been cached, made a new release it has 1.14.3 now.

fix3000, 2023-01-24:
> Good now, successfully flashed a few modules at 5Mbps on the pro unit.
> Detection and mounting are both faster than we remember for CM3.
>
> Desktop windows etcher is still not super stable but we were testing on
> some incredibly cheap and slow laptops with potentially bad usb
> controllers. I'll do some testing on regular machines next.

I can also confirm success of detecting and flashing CM4 and CM3+
(simultaneously, no less) on Etcher Pro. V1.14.3, sn S000079.

Thank you very much for getting this nailed down, my quality of life is
about to improve significantly.

mcraa, 2023-02-14:
Thanks for confirming again.

The EtcherPro specific issue could go in its repo https://github.com/balena-io-hardware/etcherPro-fleet-sw/issues
Or somewhere in the forums https://forums.balena.io/t/etcherpro-progress-updates/23741/82

closed 2023-02-14 by mcraa

timetraveler90, 2024-02-06:
I still have this issue on Windows 11, what are the options to work around it ?