balena-io/etcher #3793: Error spawn C:\Temp ENOENT

I-iunnius, 2022-08-05:
- **Etcher version:** 1.7.9
- **Operating system and architecture:** W7 Ultimate, 64 bits. SD Card 128 GB XC-I U3, new.
- **Image flashed:** motioneyeos-raspberrypi3.img.gz (GitHub) Checked by Kaspersky.
- **What do you think should have happened:** No idea, even googling; Steps: Flash from file: neyeos-raspberrypi3.img.gz + Select target: Generic- Multi-Card USB Device Size 134 GB Location F:\ Large Drive Select (1) + Flash: WARNING! ... Are you sure ...? Yes, I´m sure
- **What happened:** Attention. Something went wrong. If it is a compressed image, please check that the archive is not corrupted. spawn C:\Temp ENOENT
- **Do you see any meaningful error information in the DevTools?** Unchecked runtime.lastError: The message port closed before a response was received.
- Thank you

I-iunnius, 2022-08-07:
I have found this in https://forums.balena.io/t/balenaetcher-stopped-working/344965
from wwmckay :
...............................................

(In response to dtischler, sep 21):

Thanks for the suggestions. I was able to get it working and wanted to share how I did it.

A while back I had added some system variables that affected Balena. I removed the added variables involving the “platform-tools” and Etcher still wasn’t working; the fix was that under System Variables, ComSpec was “C:\WINDOWS\system32\cmd.exe;” and Etcher was looking for “C:\WINDOWS\system32\cmd.exe”. Once I removed the semicolon Etcher worked.

Thanks again
..............................................

Could anyone tell me how to do it?

Thank you

lurch, 2022-08-07:
> Could anyone tell me how to do it?

https://www.google.com/search?q=windows+edit+system+variables

I-iunnius, 2022-08-13:
Yes. Thank you lurch. It works:
Computer, and select Properties. Select Advanced System Settings. Select  lower System Variables (systemwide settings) lists. Select Path and Edit. At the end, add C:\WINDOWS\system32\cmd.exe
Accept, and restart.

closed 2022-08-13 by I-iunnius