balena-io/etcher #3795: Flash Failed

Zapnologica, 2022-08-06:
- **1.7.9** 
- **windows 11:** 
- **Home Assistant** 
- **It should have flashed it, but just says failed with no error:** 
- **See screen shor:**
- **Yes** 

[image]

[image]

mcraa, 2022-08-08:
Based on the screenshot I guess the error is right after you click `Flash`. Can you confirm?
Did you get the prompt for the admin password? (could appear under some other window)
If you can open the `>errors:` line in the logs or copy its contents that may contain some useful information to find out what is the problem.

LionelBrink, 2022-08-09:
I'm experiencing the same issue on Etcher 1.7.9
> Windows 11 21H2  Build 22000.832
> Attempting to flash to USB connected SD card reader/writer using a local copy of https://github.com/home-assistant/operating-system/releases/download/8.4/haos_rpi3-64-8.4.img.xz & URL
> Decompression completes & flash fails almost immediately - corrupting the SD card
> I've tried 2 different SD cards (confirmed on phone & camera prior to flash) & also confirmed the SD card writer is working (FAT32 format, read & write)
Please let me know where I can find the log file - I've searched the usual folders but cannot locate anything related to Balena-IO or Etcher

EternalNOP-412, 2022-08-19:
I have the same issue.  It erased the flash drive and I can't format the flash drive again with Windows.  Didn't make screenshots but the version is the same.

For me it was right after I hit Flash.

Zapnologica, 2022-08-19:
I think I have found the solution, Don't ask me how.

But you need to turn windows defender anti randsomware protection off.  Then the flashing will work.

[image]

zwhitchcox, 2022-08-19:
@mcraa This could be related to #829 ...it seems like Microsoft is less prone to trust things not published the the microsoft store

zwhitchcox, 2022-08-19:
I think we really need a comprehensive guide for permissions, because security is being implemented more and more at the OS level.

For instance, giving access to drives on Mac OS X, white listing Etcher for Windows "Defender",

lurch, 2022-08-20:
This has cropped up several times before https://github.com/balena-io/etcher/search?q=controlled+folder+access&type=issues

EternalNOP-412, 2022-08-20:
Well, I got it to work on my Win 10 machine that isn't on Azure AD.  The issue I had there is that Acronis asked if the program was OK, but the program ran anyways and it failed again.  But since I told Acronis it was OK, I was able to correct the USB drive and run it again and I was good.

So, is there a way to capture Anti-Virus active and create an alert to ensure the AV has the program marked safe?  Yea, I know this could be pandora's box of lots of AVs.  But Windows 10/11 seem to track who is up.

lurch, 2022-08-21:
See the comment here? https://github.com/balena-io/etcher/issues/3296#issuecomment-691891081

zwhitchcox, 2022-08-21:
Another permissions issue: https://github.com/balena-io/etcher/issues/3751#issuecomment-1221555875

Jay714, 2022-11-26:
I'm getting the flash failed error as soon as the flash has compleated and at about 15% into the valadation. I have tried all the fixes that I have been able to find and nothing works. I'm not a technical person and I attached the error report. 
[Balena Etcher Error file.docx](https://github.com/balena-io/etcher/files/10096782/Balena.Etcher.Error.file.docx)

Bob-Down, 2023-03-29:
> I think I have found the solution, Don't ask me how.
> 
> But you need to turn windows defender anti randsomware protection off. Then the flashing will work.
> 
> [image]

Zapnologica - Sir you are a star! After making 3 SD card unreadable by window (I was able to reformat them using the SD card formatter app), editing the RegFile, etc. I came across your post saying to disable the Ramson Ware checking option and it worked perfectly! One very happy camper can now continue to get Home Assistant up and running.

closed 2023-07-12 by dfunckt