balena-io/etcher #3796: "GPU process isn't usable" on Ubuntu 22.04 with NVIDIA GPU

Utopiah, 2022-08-07:
- **Etcher version:** 1.7.9-x64.AppImage
- **Operating system and architecture:** Ubuntu 22.04 x64
- **Image flashed:** none
- **What do you think should have happened:** work 
- **What happened:** failed to start
- **Do you see any meaningful error information in the DevTools?** CLI output :

`
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
[189375:0807/155832.144524:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
/tmp/.mount_balenaJR4PZL/balena-etcher-electron: line 10: 189375 Trace/breakpoint trap   (core dumped) "${script_dir}"/balena-etcher-electron.bin "$@"
`

According to https://github.com/webosbrew/dev-manager-desktop/issues/9 updating electron solves the issue.

lurch, 2022-08-07:
See also https://github.com/balena-io/etcher/search?q=GPU+process+usable&type=issues

Utopiah, 2022-08-07:
Thanks IMHO it explains why but doesn’t help at execution. An error message when access isn’t sufficient would help.

david2628, 2022-09-17:
https://github.com/balena-io/etcher/issues/3639 
there is more help :D 
or just start with `/usr/bin/balena-etcher-electron --disable-gpu-sandbox `

wes8ty, 2022-10-28:
Seeing this issue with Debian 11

```
nvidia-smi 
Fri Oct 28 14:01:24 2022       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 510.85.02    Driver Version: 510.85.02    CUDA Version: 11.6     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA T1200 La...  On   | 00000000:01:00.0  On |                  N/A |
| N/A   52C    P5    12W /  N/A |   1104MiB /  4096MiB |     11%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
```

Utopiah, 2022-10-30:
I find this surprising that somehow an "answer" is a link to numerous other issues with the same problem. Clearly the error message is pointless because it shows nothing that helps to solve it. As the feedback here seems to be "this is not an actual bug" thus that the user somehow should know I'll close the issue and stop getting notifications on this but I find this a rather counter productive for on-boarding.

closed 2022-10-30 by Utopiah