balena-io/etcher #3797: by design or error?

dledwith, 2022-08-07:
I am flashing ubuntu 22.04  to a MBR formatted flash drive and it keeps coming out GPT. Is this by design from balena or ubuntu or is this an error? My goal is the install to pre UEFI computers that don't boot from GPT. Please advise and thank you in advance.

lurch, 2022-08-08:
The partition-table used on the image you flash is entirely determined by the image. So if the Ubuntu image you're flashing is "coming out" as GPT, then it means that Ubuntu disk image is using GPT. As the partition-table is just contained at the start of the disk image.
See also https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos

The Rufus or UNetbootin tools mentioned in https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md _might_ let you create an MBR-partitioned flash drive, as they work in a different way to Etcher.

(BTW I'm not an Etcher developer, just a helpful community member :slightly_smiling_face: )

dledwith, 2023-02-05:
Yeah Rufus did the trick, thanks.

closed 2023-02-05 by dledwith