balena-io/etcher #3798: Does not work on OSX with non admin user

gradha, 2022-08-08:
- **Etcher version:** 1.7.9
- **Operating system and architecture:** osx monterey 12.4.
- **Image flashed:** Clonezila
- **What do you think should have happened:** Flash
- **What happened:** Nothing.
- **Do you see any meaningful error information in the DevTools?** 

When the dialog opens asking for permission, it doesn't allow one to specify a different user than the current one. Thus, if you are trying to flash something with a non admin user, even if you know the user/pass of the admin account the software won't do anything.

[image]

thedocbwarren, 2022-08-08:
macOS in general will always work this way from here on.  This is a feature of the OS to manage access to devices.  Writing to a device is privileged access.

gradha, 2022-08-09:
> macOS in general will always work this way from here on. This is a feature of the OS to manage access to devices. Writing to a device is privileged access.

And that's why macOS in general asks you for the user name of the admin account and does not presume you are on. Or at least that's how it's been for the last 15 years I've been using it as a non admin user.

maclm, 2022-09-06:
A common example of expected behaviour is updating an app from a standard (non-admin) account: macOS will prompt for username & password.
My hunch is _balenaEtcher_ asks the system for drive access, while it should ask for admin permissions.

dfunckt, 2023-07-12:
Closing in favor of #3321

closed 2023-07-12 by dfunckt