balena-io/etcher #3800: Flash failed! - false

TomaxMe, 2022-08-12:
- **1.7.9 version:** 
- **Win11, 64-bit:** 
- **Openhab - https://www.openhab.org/download/** 
- **What do you think should have happened:** 
- **Flash failed! error displayed when is checking the flashed USB-stick (32 Gb capacity). Tried multiple times. The good news is that is a false alarm since the image si correctly flashed on the USB-stick. **

mcraa, 2022-08-15:
Thanks for sharing,
most of the time it is just the unmount of the device that fails at the end.
However, it can be multiple other things while the flashing is still successful.
We'll work on making the messages more explicit.

duplicate of #3730

closed 2022-08-15 by mcraa

lurch, 2022-08-15:
> The good news is that is a false alarm since the image si correctly flashed on the USB-stick.

There's also the (slight?) possibility that there was _actually_ data-corruption on your USB-stick (e.g. due to faulty or fake hardware), which might cause the "Flash failed" message because verification failed; but this corruption might not be enough to cause the boot to actually fail, e.g. if the corruption is in a file not read during the boot process? :shrug: