balena-io/etcher #3801: Error - E: Unable to locate package balena-etcher-electron. How do I fix this?!

DMan-Jam, 2022-08-13:
Etcher version: 1.7.9
- Linux Mint 21 Cinnamon, amd64

I have no idea why it isn't installing properly. I've looked up videos, attempted installation thru command line multiple ways, deleted balena files over and over before reinstalling (including coredump) and it keeps giving me the same errors. I've tried all the tips and tricks in root as well; nothing works. :(
[image]

lurch, 2022-08-13:
I think if things had been [set up](https://github.com/balena-io/etcher/issues/3501) properly, you ought to also have `https://dl.cloudsmith.io/...` in the list of repositories reported by `apt-get update`? :man_shrugging:

mauro-miatello, 2022-11-10:
In my Mint 21 the process goes forward but hangs with:
```
chmod: cannot access '/opt/balena-etcher-electron/chrome-sandbox': No such file or directory
```
so I create a symlink
```
cd /opt
sudo ln -s balenaEtcher/balena-etcher-electron
```
and then
```
sudo apt-get install --reinstall balena-etcher-electron
```
that completed without errors, but when I launch Balena nothing happens

Jeroenus76, 2023-02-13:
i have the same problems, but then on my raspberry pi running debian bullseye.

jeroen@rpi4:~ $ curl -1sLf \
      'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh' \
      | sudo -E bash
Executing the  setup script for the 'balena/etcher' repository ...

   OK: Checking for required executable 'curl' ...
   OK: Checking for required executable 'apt-get' ...
   OK: Detecting your OS distribution and release using system methods ...
 ^^^^: ... Detected/provided for your OS/distribution, version and architecture:
 >>>>:
 >>>>: ... distro=debian  version=11  codename=bullseye  arch=aarch64  
 >>>>:
   OK: Checking for apt dependency 'apt-transport-https' ...
   OK: Checking for apt dependency 'ca-certificates' ...
   OK: Checking for apt dependency 'gnupg' ...
   OK: Checking for apt signed-by key support ...
   OK: Importing 'balena/etcher' repository GPG key ...
   OK: Checking if upstream install config is OK ...
   OK: Installing 'balena/etcher' repository via apt ...
   OK: Updating apt repository metadata cache ...
   OK: The repository has been installed successfully - You're ready to rock!

jeroen@rpi4:~ $ sudo apt install -y balena-etcher-electron
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package balena-etcher-electron

mcraa, 2023-02-13:
Since our CI update the package name changed to `balena-etcher` (no electron at the end).
The `balena-etcher-electron` package is still available on cloudsmith with version 1.7.9, so it is unclear why is it not found.
If you want to install 1.13.1 you may try `sudo apt install -y balena-etcher`.
We will investigate and update the repo and docs accordingly.