balena-io/etcher #3806: Error: Couldn’t clean the drive, Command failed: diskpart /s

Konglomneshued, 2022-08-24:
- **Etcher version: 1.7.9** 
- **Operating system and architecture: Windows 11 Home** 
- **Image flashed: pop-os_22.04_amd64_nvidia_12** 
- **What do you think should have happened: Flash failed.  Yes I expected it to fail, only had problems with this software.** 
- **What happened: Flash failed as expected although my USB is not bricked, so that's good.**
- **Do you see any meaningful error information in the DevTools?** 

I posted the error as the title.  I saw someone say this kind of issue is related to https://github.com/balena-io/etcher/issues/1793 , however I'm not using an SD nor an SD card hub.  Just a plain USB stick.  Not to mention that thread is still open and unresolved, with the solutions being related to SD cards.

I thought this might help too:

`({"image":{"size":3573547008,"name":"pop-os_22.04_amd64_nvidia_12.iso","hasMBR":true,"partitions":[{"offset":247808,"size":4194304,"type":239,"index":1}],"extension":"iso","path":"C:\\Users\\untitled(1)\\Downloads\\pop-os_22.04_amd64_nvidia_12.iso"},"drives":[{"size":64160400896,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"\\\\.\\PhysicalDrive1","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive1","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[],"description":"Samsung Type-C USB Device","isUAS":false,"partitionTableType":"mbr"}],"driveCount":1,"uuid":"a2f4016d-8daf-4908-a282-6432b494baa5","status":"finished","flashInstanceUuid":"a2f4016d-8daf-4908-a282-6432b494baa5","errors":[{"name":"Error","description":"Samsung Type-C USB Device","device":"\\\\.\\PhysicalDrive1"}],"devices":{"failed":1,"successful":0},"bytesWritten":0,"sourceMetadata":{"size":3573547008,"name":"pop-os_22.04_amd64_nvidia_12.iso","blocks":[{"blocks":[{"offset":0,"length":3573547008}]}],"blockmappedSize":3573547008},"sample":0.1,"applicationSessionUuid":"75b4100e-fa5a-4047-8d9d-1f373fccd3c0","flashingWorkflowUuid":"f7aca131-1139-4ff2-afe3-709fd8bc6839"})`

I have installed Etcher non-portable and run it as admin after a fresh restart.  My USB is completely unallocated and I verified the checksum of the iso I downloaded.

lurch, 2022-08-24:
> What do you think should have happened: Flash failed. Yes I expected it to fail

:rofl:

zwhitchcox, 2022-08-25:
>I saw someone say this kind of issue is related to https://github.com/balena-io/etcher/issues/1793 , however I'm not using an SD nor an SD card hub. Just a plain USB stick.

Yes, this as unlikely, because USB sticks do not have a locking functionality.

Can you try disabling Windows Defender?

Konglomneshued, 2022-08-25:
> > I saw someone say this kind of issue is related to #1793 , however I'm not using an SD nor an SD card hub. Just a plain USB stick.
> 
> Yes, this as unlikely, because USB sticks do not have a locking functionality.
> 
> Can you try disabling Windows Defender?

This worked, thanks.

closed 2022-08-25 by Konglomneshued