balena-io/etcher #3807: Feature request: Read SD card contents back to files

Krellan, 2022-08-27:
This isn't a bug, but a feature request.

I would love to see a feature to run the etching process in reverse. That is, read the full SD card contents back to a file. Then, I could read in an entire SD card and easily have a backup of its contents. I could then use the image to etch another card, to make a byte-for-byte exact copy of it.

Anybody else think this would be a good feature to have?

lurch, 2022-08-30:
Duplicate of #266 ?

Krellan, 2022-08-30:
Thanks. #266 is of greater scope, performing a backup of a drive. I'm thinking of something of lesser scope, just the ability to read the entire contents of the drive into a file. Perhaps a subset of #266.

The ability to perform a backup can come after that (since, as #266 says, there's lots of issues regarding differently-sized SD cards, wasted padding space after the end of the partitions, and so forth).