balena-io/etcher #3809: Etcher won't start

JoseBCN84, 2022-09-06:
- **Etcher version: **balena-etcher-electron-1.7.7-1.x86_64 
- **Operating system and architecture: ** Fedora 36 64-bit, Gnome 42.4, X11, with NVIDIA GeForce GTX 1070
- **What do you think should have happened:** I'm just expecting to start the app
- **What happened:** I downloaded the appimage file. But it doesn't start. 
I deleted the appimage file and installed etcher using dnf on terminal. installation was OK but still when I try to start the app nothing happens.  

- **Do you see any meaningful error information in the DevTools?** 
- When I tried to run it via terminal I get the following:
```
[Axios v1.7.9] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'silentJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
[4747:0906/092554.774678:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
/tmp/.mount_balenagyTq8j/balena-etcher-electron: line 10:  4747 Trace/breakpoint trap   (core dumped) "${script_dir}"/balena-etcher-electron.bin "$@"
```
Any advice?

Thanks!

lurch, 2022-09-06:
See also https://github.com/balena-io/etcher/search?q=GPU+process+usable&type=issues

david2628, 2022-09-17:
try `/usr/bin/balena-etcher-electron --disable-gpu-sandbox` in your console

JoseBCN84, 2022-09-18:
Thanks @david2628, tried that but I got the below but I got `No such file or directory` althought if I navigate via Files, I can see it...

david2628, 2022-09-18:
you got it running in the terminal earlier.
just add the flag `--disable-gpu-sandbox`

JoseBCN84, 2022-09-18:
That's the thing, I can see the file, if I grafically navigate to the folder. but when I do it in the terminal it tells me it doesn't exist... I'm confused

JoseBCN84, 2022-09-18:
OK, I figured out what happened. Somehow, the file `/usr/bin/balena-etcher-electron` was a broken symbolic link.
I found the actual file `/usr/bin/balena-etcher-electron` so I deleted the broken symbolic link and created a new one in `usr/bin/`, ran it adding `--disable-gpu-sandbox` as you suggested and it worked.
Thanks!

Would there be a way to run it without having to use the terminal?

henning, 2022-09-18:
@JoseBCN84 if you need to do such fiddling with symlinks in /usr/bin this might also be worth a bug report to the developers who make the package you use - this shouldn't happen :)

JoseBCN84, 2022-09-18:
I’ve installed and uninstalled it so many times lately trying to get it to work that I wouldn’t be surprised I broke something tbh

david2628, 2022-09-18:
there is something explained here https://github.com/balena-io/etcher/issues/3639#issuecomment-1153285345
but i have not tried it yet so take it with a grain of salt

lurch, 2022-09-19:
> if you need to do such fiddling with symlinks in /usr/bin this might also be worth a bug report to the developers

There's already a separate issue for that :wink: #3771

peliopoulos, 2022-12-14:
Fresh install of Ubuntu 22.04 here with an nVidia card - the following command worked:

`/opt/balenaEtcher/balena-etcher-electron --disable-gpu-sandbox`

peliopoulos, 2022-12-14:
O/S: Ubuntu 22.04
GPU: Nvidia
Etcher version: 1.10.6

changed the Desktop icon file to add the --disable-gpu-sandbox option

`sudo gedit /usr/share/applications/balena-etcher-electron.desktop`

changed the Exec line to read:

`Exec=/opt/balenaEtcher/balena-etcher-electron --disable-gpu-sandbox %U`

rebooted and now it works great!

linuxlite, 2022-12-28:
Still not working in balena-etcher_1.12.7_amd64.deb

closed 2024-05-14 by JoseBCN84