balena-io/etcher #3812: balea Etcher 1.7.9 - stucks on "Finishing"

JOLOalpha, 2022-09-15:
- balenaEtcher 1.7.9
- Windows 10
- SIMATIC_IPC_Image_PartitionCreator_V3.6.2.img 

Hi,
I started flashing an image file to a USB stick, solidly works, then stucks in state "Finishing" since allready 1 hour.
Could you kindly help me?

mcraa, 2022-09-15:
Did you check the contents of the USB stick? It could be done already.
Can you open the devtools (Ctrl + Shift + I) and check what logs are there?

JOLOalpha, 2022-09-15:
Hi,

thanks a lot for your feedback. I canceled the action „Finishing“, took a shot oft the files (attached), formated the USB stick and started the flashing meanwhile again.

The iso files name contains 3 dots. Could this be an issue? -> SIMATIC_IPC_Image_PartitionCreator_V3.6.2.img

***@***.***

Freundliche Grüße / Best regards
Tobias Rademacher

***@***.***

TOBIAS RADEMACHER
AUTOMATION ENGINEER
An der Garnbleiche 4
52349 Düren
T +49 2421 307 570
M +49 173 5478 032
***@***.******@***.***>
www.tunnelservicegroup.de<http://www.tunnelservicegroup.de/>

JOLOalpha, 2022-09-15:
Hi,

stucking in „Finishing“ again. Here we have the part of the logg with errors:

***@***.***

The logg ends with:

***@***.***

Should this image be okay?

Thanks before!

Freundliche Grüße / Best regards
Tobias Rademacher

***@***.***

TOBIAS RADEMACHER
AUTOMATION ENGINEER
An der Garnbleiche 4
52349 Düren
T +49 2421 307 570
M +49 173 5478 032
***@***.******@***.***>
www.tunnelservicegroup.de<http://www.tunnelservicegroup.de/>

mcraa, 2022-09-15:
I think not all is visible in the github issue of what you send in email.

lurch, 2022-09-15:
Yeah, replying to a GitHub issue via email ignores any attachments / images / etc. AFAIK you can **only** add attachments to a Github issue by visiting the GH website (so in this case https://github.com/balena-io/etcher/issues/3812 ).

JOLOalpha, 2022-10-11:
Hi,

yeah, this is not the complete logg.

I only sent you two shots - the parts marked as warning and the finish.

Freundliche Grüße / Best regards
Tobias Rademacher

***@***.***

TOBIAS RADEMACHER
AUTOMATION ENGINEER
An der Garnbleiche 4
52349 Düren
T +49 2421 307 570
M +49 173 5478 032
***@***.******@***.***>
www.tunnelservicegroup.de<http://www.tunnelservicegroup.de/>

lurch, 2022-10-11:
@JOLOalpha Please look at the actual GitHub website, and you'll see that your images haven't appeared there.

closed 2023-07-12 by dfunckt