balena-io/etcher #3814: convert or option to flash as GPT to MBR partition

cgvirus, 2022-09-17:
The standard flash is always GPT in Linux now. 
But sometimes older machine requires MBR. A conversion or option to flash the drive as MBR or GPT would be great.
https://askubuntu.com/questions/1288476/can-not-install-from-usb-with-error-cannot-find-grub-platform

dutandrewkulang, 2022-10-18:
yah that would be a good ideal to have option to select the flash option for the computer.

zwhitchcox, 2022-10-19:
Hi, the good news is we are working on tools that will enable this right now, e.g. [balena-parted](https://github.com/zwhitchcox/balena-parted). However, there's more to it than just changing the partition type. I think this would probably be better as plug-in functionality, as opposed to a built in feature. I'll leave this open for a future plug-in idea though.

anasouardini, 2023-09-19:
God bless binbows tools (Rufus).