balena-io/etcher #3815: Etcher won't start

david2628, 2022-09-17:
- **Etcher version:** 1.7.9
- **Operating system and architecture:** Arch-Linux 5.19.8-arch1-1
- **Image flashed:** none
- **What do you think should have happened:** Etcher should start 
- **Do you see any meaningful error information in the DevTools?**
Starting it with zsh via ` usr/bin/balena-etcher-electron `  shows: 
```
[Axios v1.7.9] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'silentJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[70105:0917/144657.444056:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
zsh: trace trap (core dumped)  /usr/bin/balena-etcher-electron
```
by adding `--disable-gpu-sandbox` i can start it.
Please add that flag
< !-- You can open DevTools by pressing `Ctrl+Shift+I` (`Ctrl+Alt+I` for Etcher before v1.3.x), or `Cmd+Opt+I` if you're on macOS. -->

henning, 2022-09-18:
I just experienced the same on UBuntu 22.04:
```

$ ./balenaEtcher-1.7.9-x64.AppImage 
[Axios v1.7.9] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'silentJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[106698:0918/165557.780990:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
/tmp/.mount_balena5t3SeL/balena-etcher-electron: line 10: 106698 Trace/breakpoint trap   (core dumped) "${script_dir}"/balena-etcher-electron.bin "$@"

```

would be great to have this documented somewhere in a prominent place when downloading, so people see it when downloading the app, if it's not possible to make the script just work for all without adding that flag manually. ;)

I also got an nvidia card.

patrickwasp, 2022-09-19:
also, I had to run `sudo apt install libfuse2` to get the app image to work. I think an official snap with all the correct settings would make things simpler to use.

Ccode-lang, 2022-10-05:
libfuse is for normal app launch.  non-stripped ubuntu should launch correctly @patrickwasp.

NicTanghe, 2022-10-31:
whould be nice if dependancies where listed

lurch, 2022-11-03:
> whould be nice if dependancies where listed

https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#runtime-gnulinux-dependencies (although I dunno if that list is still up-to-date or not?)

zwhitchcox, 2022-11-03:
Those should be up to date.

crowetic, 2022-11-14:
I just 

> I just experienced the same on UBuntu 22.04:
>
> ```
>
> $ ./balenaEtcher-1.7.9-x64.AppImage
> [Axios v1.7.9] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed [...]

You can start it with 

```./balenaEtcher-1.7.9-x64.AppImage --no-sandbox``` 

until the devs get around to actually fixing this.